tpaviot/pythonocc-core #174: PyQt5 support

jf---, 2015-12-23:
with Qt4 reaching no longer being supported [post 19th of Dec 2015])(http://blog.qt.io/blog/2014/11/27/qt-4-8-x-support-to-be-extended-for-another-year/), its time to get going on providing support for Qt5 / PyQt5. 

finally, with Qt5, QML is really becoming to way to do GUI's in Qt, so this wedges into supporting some #173 / QML examples

rainman110, 2015-12-28:
Please not, that OpenCASCADE does not fully support QT5 yet. There are problems with MDI areas and the 3D visualization. This will be fixed in 7.0 though. We can back-port the 7.0 fix to OCE if we want.

jf---, 2015-12-31:
Hi @rainman110 , thanks for pointing this out... :+1: 
Yes, I think that back-porting that patch is an excellent idea...

Migrating to 7.0.+ will take some time so merging this into OCE 0.18 ( -> OCC 6.9.1 ) decoupling Qt5 & OCE 0.18+ makes makes a lot of sense to me... 

Which patches are you referring to exactly?
I've tried to grep the [commit log like such](https://www.google.nl/search?q=site%3Agit.dev.opencascade.org+qt5&oq=site%3Agit.dev.opencascade.org+qt5&aqs=chrome..69i57j69i58.5703j0j4&sourceid=chrome&es_sm=91&ie=UTF-8#q=site:git.dev.opencascade.org+qt+mdi) but couldnt find the specific commit you're mentioning...

rainman110, 2015-12-31:
Hi @jf--- I am referring to the issue http://tracker.dev.opencascade.org/view.php?id=26603.

The file OpenGl_Window.cxx needs to be patched according to this change:
http://git.dev.opencascade.org/gitweb/?p=occt.git;a=commitdiff;h=62e1beed962f76420131e79de4ffa8748e70e50d#patch26

Also, we should incorporate the qt specific occt window class, which fixes the error finally:
 - [OcctWindow.cxx](http://git.dev.opencascade.org/gitweb/?p=occt.git;a=blob;f=samples/qt/Common/src/OcctWindow.cxx;h=c9bc3cad440bad51a682825d7b1b0113b28bd49e;hb=de6273a31f56442d7073d4de974aef5d055e19e7)
 - [OcctWindow.hxx](http://git.dev.opencascade.org/gitweb/?p=occt.git;a=blob;f=samples/qt/Common/src/OcctWindow.h;h=80c9668ee86c78a0c1bbf61d495c141c66bfa3ca;hb=de6273a31f56442d7073d4de974aef5d055e19e7)

I tried both in a separate project and it works like a charm :grinning:

jf---, 2016-01-03:
@rainman110 , thanks for those pointers... that's critical information...

Yes, I think it would be wise of for us to integrate these patches...
I take it these are merged in OCC 7.0.0 not 6.9.+?

rainman110, 2016-01-05:
These commits are part of the upcoming OCCT 7.0.0 release. Still no problem in integrate it into OCE before.

jf---, 2016-01-05:
yes, shall we take this over at the OCE repo?

rainman110, 2016-01-05:
Yes, this issue belongs to OCE.

rainman110, 2016-01-06:
@jf--- Actually, this issue belongs to both bug trackers.

 1. The back-porting of OpenGl-Window is OCE specific
 2. Integration of OcctWindow must be done in [pythonocc-core / Display3d.cpp](https://github.com/tpaviot/pythonocc-core/blob/master/src/Visualization/Display3d.cpp)

rainman110, 2016-01-11:
Hi @jf---! I tested the Qt5 code. I've seen two issues:

   1. Crash due to non existing ```QWheelEvent::delta``` method. This method is deprecated in Qt5 and was renamed to ```QWheelEvent::angleDelta```. Crashes only in case of a wheel event obviously.
   2. Many warnings like:

         QWindowsGLContext::swapBuffers: Cannot find window 0x11048e

Except from that, it works quite well :grinning:

jf---, 2016-01-11:
Ai, @rainman110 , that's great you found this...

Damn, the wheelevent was solved in the overpaint example, but not in qtBaseViewer.py. 
I [just fixed this](https://github.com/tpaviot/pythonocc-core/pull/196), thanks so much for noticing this... :+1: 

Wrt swapbuffers; overpainting the viewer with pyqt5 isnt as polished as with pyqt4, correct.
You can see this when zooming a rectangle ( Shift + RMB ) 
However, when I do so ( PyQt5 / Qt5.5 ) I dont have the swapbuffers warning.
Which version of pyqt are you using on which platform?

rainman110, 2016-01-11:
@jf---  I am using qt/pyqt 5.5.1 win64, python 3.5 from conda. 

I took qt5 from the conda channel  https://anaconda.org/mmcauliffe

jf---, 2016-01-11:
Thanks for the feedback @rainman110 ...

tpaviot, 2016-01-21:
Fixed.

closed 2016-01-21 by tpaviot