tpaviot/pythonocc-core #177: Some tests or samples do not work.

WBB-Badtaste, 2015-12-28:
After installing the pythonocc-core I try to run some samples but they don't work.

>wubinbindeMacBook-Air:test Martin$ python /Users/Martin/Documents/pythonocc-core/examples/core_topology_prism.py 
PyQt4 not found - using PySide
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
Cocoa window created.
V3d_Viewer created.
V3d_View created
AIS_InteractiveContext created.
Display3d class successfully initialized.
 ########################################
Segmentation fault: 11

Then I try to run the uint tests, I found the last one crash.

>wubinbindeMacBook-Air:test Martin$ python run_tests.py
testAutoImportOfDependentModules (core_wrapper_features_unittest.TestWrapperFeatures)
Test: automatic import of dependent modules ... ok
testProtectedConstructor (core_wrapper_features_unittest.TestWrapperFeatures)
Test: protected constructor ... ok
testStandardBooleanByRefPassedReturned (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
testTopoDS_byref_arguments (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_const_Standard_Integer_byref (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_const_Standard_Real_byref (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_default_constructor_DEFINE_STANDARD_ALLOC (core_wrapper_features_unittest.TestWrapperFeatures)
OCE classes the defines standard alllocator can be instanciated ... ok
test_dict (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_dump_to_string (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_eq_operator (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_ft1 (core_wrapper_features_unittest.TestWrapperFeatures)
Test: Standard_Integer & by reference transformator ... ok
test_hash (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_hash_eq_operator (core_wrapper_features_unittest.TestWrapperFeatures)
test that the == wrapper is ok ... ok
test_inherit_topods_shape (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_list (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_neq_operator (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_pickle_from_file (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_standard_boolean_byref (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_standard_integer_by_ref_passed_returned (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_static_method (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_sub_class (core_wrapper_features_unittest.TestWrapperFeatures)
Test: subclass ... ok
test_topology (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_traverse_box_topology (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_axis (core_geometry_unittest.TestGeometry)
Test: axis ... ok
test_bezier_surfaces (core_geometry_unittest.TestGeometry)
Test: Bezier surfaces ... Segmentation fault: 11

How can I fixed it？

tpaviot, 2015-12-28:
Is the core_helloworld.py example pass ?

WBB-Badtaste, 2015-12-28:
Yes, it work.

wubinbindeMacBook-Air:getting_started Martin$ python
/Users/Martin/Documents/pythonocc-core/examples/core_helloworld.py

PyQt4 not found - using PySide

 ###### 3D rendering pipe initialisation #####

Display3d class initialization starting ...

Aspect_DisplayConnection created.

Graphic_Driver created.

Cocoa window created.

V3d_Viewer created.

V3d_View created

AIS_InteractiveContext created.

Display3d class successfully initialized.

 ########################################
[image: 内嵌图片 1]

Martin Wu
//Shenzhen Sunet Industry co.,Ltd.
//E-mail:sos901012@gmail.com <sos901012@gmail.com>
//TEL:+86-18576761645
//QQ:705768492

tpaviot, 2015-12-28:
ok thank you. What's your os version please ? c compiler ?

WBB-Badtaste, 2015-12-28:
os x 10.10
clang

Martin Wu
//Shenzhen Sunet Industry co.,Ltd.
//E-mail:sos901012@gmail.com <sos901012@gmail.com>
//TEL:+86-18576761645
//QQ:705768492

WBB-Badtaste, 2015-12-29:
[image: 内嵌图片 1]

Martin Wu
//Shenzhen Sunet Industry co.,Ltd.
//E-mail:sos901012@gmail.com <sos901012@gmail.com>
//TEL:+86-18576761645
//QQ:705768492

2015-12-28 17:49 GMT+08:00 吴彬彬 <sos901012@gmail.com>:

rainman110, 2016-01-13:
I had the same issue on my mac

WBB-Badtaste, 2016-01-13:
I fix it by using conda to install the occ.

$ conda install -c https://conda.anaconda.org/dlr-sc pythonocc-core

rainman110, 2016-01-13:
Did the unit tests run without crash? We have a problem with the test_bezier_surfaces test.

WBB-Badtaste, 2016-01-14:
I can't find this test-file

[image]

tpaviot, 2016-01-14:
@WBB-Badtaste Just run:
```bash
$ python run_tests.py
```

WBB-Badtaste, 2016-01-15:
It's ok

Martin Wu
//Shenzhen Sunet Industry co.,Ltd.
//E-mail:sos901012@gmail.com <sos901012@gmail.com>
//TEL:+86-18576761645
//QQ:705768492

WBB-Badtaste, 2016-01-18:
吴彬彬邀请您加入全球最大的职业社交网站 LinkedIn (领英)。领英能帮助您管理职业生涯，让您接触到全球各地的工作机会。 快来加入领英，创建您的职业资料，轻松拓展职场人脉！

吴彬彬
软件工程师 — 深圳相聚实业有限公司
中国 广东 深圳

接受邀请: https://www.linkedin.com/e/v2?e=0-ijkbmzsx-di&t=ssuw&tracking=eml-china-m2g-a-cta&ek=invite_guest&sharedKey=M0vHaj_m&invitationId=6094922111070068737

有会员邀请您建立联系。在“猜您认识”等功能中，领英将使用您的邮箱地址来向会员推荐人脉。在此处退订: https://www.linkedin.com/e/v2?e=0-ijkbmzsx-di&t=lun&midToken=AQFmrFVhGEd36Q&ek=invite_guest&loid=AQGEkch_SG3ErwAAAVJWDdbL2eNpzKRuZzMEgl4aJxhA96dxT7gtXCV7NyVw7oTXsXs_HpXe_YcFadRfw1XxbTaGRiA6r4p8Kof0A6WVKIKHz_OeuCNZ8hxfYXttscundjkWDtXf5ua0t4krEll282LPIs-R4JSqZzuRFkVvMigN-yI7J7XymNndbF87i0GCz7MC&eid=0-ijkbmzsx-di

此邮件的收件人是reply@reply.github.com。

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=0-ijkbmzsx-di&a=customerServiceUrl&ek=invite_guest

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043.LinkedIn 和 LinkedIn 标志是领英的注册商标。

WBB-Badtaste, 2016-01-18:
吴彬彬邀请您加入全球最大的职业社交网站 LinkedIn (领英)。领英能帮助您管理职业生涯，让您接触到全球各地的工作机会。 快来加入领英，创建您的职业资料，轻松拓展职场人脉！

吴彬彬
软件工程师 — 深圳相聚实业有限公司
中国 广东 深圳

接受邀请: https://www.linkedin.com/e/v2?e=0-ijkbn07a-vk&t=ssuw&tracking=eml-china-m2g-a-cta&ek=invite_guest&sharedKey=djDG0LvT&invitationId=6094922110931656704

有会员邀请您建立联系。在“猜您认识”等功能中，领英将使用您的邮箱地址来向会员推荐人脉。在此处退订: https://www.linkedin.com/e/v2?e=0-ijkbn07a-vk&t=lun&midToken=AQFmrFVhGEd36Q&ek=invite_guest&loid=AQHcsGhEDuqoEwAAAVJWDzvCiOfs8pEHSZG8v0xLc4mnSKgjIU9B0X7Jcs8Ucgysl1NStDJOMDwsHBvHvC6a-84_GYiSWnEfpc6iQ5x8a5w5pwxgWSq1M2stJO_aZqUZx3eWgWJDV8AMDVnZo0xGVY0yYLya1FKsY0n-cYk_bRPvRf0z1kxSVe-KSD8Wrm0Z7fLC&eid=0-ijkbn07a-vk

此邮件的收件人是reply@reply.github.com。

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=0-ijkbn07a-vk&a=customerServiceUrl&ek=invite_guest

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043.LinkedIn 和 LinkedIn 标志是领英的注册商标。

WBB-Badtaste, 2016-01-18:
吴彬彬邀请您加入全球最大的职业社交网站 LinkedIn (领英)。领英能帮助您管理职业生涯，让您接触到全球各地的工作机会。 快来加入领英，创建您的职业资料，轻松拓展职场人脉！

吴彬彬
软件工程师 — 深圳相聚实业有限公司
中国 广东 深圳

接受邀请: https://www.linkedin.com/e/v2?e=0-ijkbn0gk-pl&t=ssuw&tracking=eml-china-m2g-a-cta&ek=invite_guest&sharedKey=7zD6H7T1&invitationId=6094922110776463360

有会员邀请您建立联系。在“猜您认识”等功能中，领英将使用您的邮箱地址来向会员推荐人脉。在此处退订: https://www.linkedin.com/e/v2?e=0-ijkbn0gk-pl&t=lun&midToken=AQFmrFVhGEd36Q&ek=invite_guest&loid=AQE-Hzs1Qcp7YgAAAVJWENTpVCjZrq1fkYW7Lf3Y_Ge2e0Z8d0VB6RByGylswDhYlGJC36bTXLPZG2DdS40axFmaKpeAbS8yXiobxfsL5KQp2mt38A4jOUNBOnJtMyeZwuQpTE63UoKROgR4HPaCPb8B02HT5HvbTIJgVQISF3xVh_i8c7qNp6ZYXIimeOJ1278Q&eid=0-ijkbn0gk-pl

此邮件的收件人是reply@reply.github.com。

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=0-ijkbn0gk-pl&a=customerServiceUrl&ek=invite_guest

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043.LinkedIn 和 LinkedIn 标志是领英的注册商标。

jf---, 2016-04-08:
spam lets close this isssue @tpaviot

closed 2016-04-27 by tpaviot

WBB-Badtaste, 2016-05-03:
Take the world's best courses, online. Coursera envisions a world where 
anyone, anywhere can transform their life by accessing the world’s best 
learning experience.
[Ready to learn something new?]
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3d3dy5jb3Vyc2VyYS5vcmcvP3V0bV9tZWRpdW09ZW1haWwmdXRtX3NvdXJjZT1pbnZpdGF0aW9ucyZ1dG1fY2FtcGFpZ249bm90aWZpY2F0aW9ucy5hdXRvLmNWSTk2aERoRWVhd3ZBb2tpU0ZDc3ciLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Y2RiN2IxMjYxM2I3NzMzMjAzZWU0YzQ3NzNhNDYxZTUyOThjNzc4OWE5MmNmMDAwMDAwMDExMjk4YTY1YzkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1ZJOTZoRGhFZWF3dkFva2lTRkNzdyJ9fSwidXNlcklkIjotMX0.-5Wh-3Sp2cVMLhfzubTGOTukhwTFRWlZEy0gq9lL-jA> 	
My Courses
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3d3dy5jb3Vyc2VyYS5vcmcvP3V0bV9tZWRpdW09ZW1haWwmdXRtX3NvdXJjZT1pbnZpdGF0aW9ucyZ1dG1fY2FtcGFpZ249bm90aWZpY2F0aW9ucy5hdXRvLmNWSTk2aERoRWVhd3ZBb2tpU0ZDc3ciLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Y2RiN2IxMjYxM2I3NzMzMjAzZWU0YzQ3NzNhNDYxZTUyOThjNzc4OWE5MmNmMDAwMDAwMDExMjk4YTY1YzkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1ZJOTZoRGhFZWF3dkFva2lTRkNzdyJ9fSwidXNlcklkIjotMX0.-5Wh-3Sp2cVMLhfzubTGOTukhwTFRWlZEy0gq9lL-jA> 
Course Catalog
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3d3dy5jb3Vyc2VyYS5vcmcvY291cnNlcz91dG1fbWVkaXVtPWVtYWlsJnV0bV9zb3VyY2U9aW52aXRhdGlvbnMmdXRtX2NhbXBhaWduPW5vdGlmaWNhdGlvbnMuYXV0by5jVkk5NmhEaEVlYXd2QW9raVNGQ3N3IiwidHJhY2tpbmciOnsidXNlcklkIjotMSwidXNlckVtYWlsIjoicmVwbHkrMDA4ZDMyNGNkYjdiMTI2MTNiNzczMzIwM2VlNGM0NzczYTQ2MWU1Mjk4Yzc3ODlhOTJjZjAwMDAwMDAxMTI5OGE2NWM5MmExNjljZTA3NjQ3ODc0QHJlcGx5LmdpdGh1Yi5jb20iLCJub3RpZmljYXRpb25UeXBlIjoiZ3Jvd3RoLmludml0YXRpb24iLCJjYW1wYWlnbklkIjoibm90aWZpY2F0aW9ucy5hdXRvLmNWSTk2aERoRWVhd3ZBb2tpU0ZDc3cifX0sInVzZXJJZCI6LTF9.IuY0N60j62fPtXGLOElmkkKGoP81DcEGY9cMaEftRlI>

吴彬彬 thinks you'll love learning on Coursera

We envision a world where anyone, anywhere can transform their life by 
accessing the world’s best learning experience. We offer hundreds open 
online courses, in dozens of subject areas, all from top universities and 
institutions.

Course topics range from business and management, to computer science, to 
medicine, music, education, and more.

Millions of our learners have advanced their careers, discovered new 
passions, and built lasting friendships through their Coursera courses. We 
hope you'll join us in learning without limits!

The Coursera Team

Explore Coursera
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3d3dy5jb3Vyc2VyYS5vcmcvb25ib2FyZGluZy8_dG9rZW49ZDNmNmQwYjdmZmVhYmJiNTdkNTZkMWEzMGEzNmFiN2UmZW1haWw9cmVwbHkrMDA4ZDMyNGNkYjdiMTI2MTNiNzczMzIwM2VlNGM0NzczYTQ2MWU1Mjk4Yzc3ODlhOTJjZjAwMDAwMDAxMTI5OGE2NWM5MmExNjljZTA3NjQ3ODc0QHJlcGx5LmdpdGh1Yi5jb20mbmFtZT10cGF2aW90L3B5dGhvbm9jYy1jb3JlP3V0bV9tZWRpdW09ZW1haWwmdXRtX3NvdXJjZT1pbnZpdGF0aW9ucyZ1dG1fY2FtcGFpZ249bm90aWZpY2F0aW9ucy5hdXRvLmNWSTk2aERoRWVhd3ZBb2tpU0ZDc3ciLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Y2RiN2IxMjYxM2I3NzMzMjAzZWU0YzQ3NzNhNDYxZTUyOThjNzc4OWE5MmNmMDAwMDAwMDExMjk4YTY1YzkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1ZJOTZoRGhFZWF3dkFva2lTRkNzdyJ9fSwidXNlcklkIjotMX0.MQZCuJpP7POGGqEFGKxdqO2MGmNeWatSAbGdl1VkB_E>
Download our App:  iOS
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL2l0dW5lcy5hcHBsZS5jb20vYXBwL2FwcGxlLXN0b3JlL2lkNzM2NTM1OTYxP2N0PU1vYmlsZSUyMFByb21vJTIwRW1haWwmdXRtX3NvdXJjZT1pbnZpdGF0aW9ucyZtdD04JnB0PTIzMzQxNTAmdXRtX2NhbXBhaWduPW5vdGlmaWNhdGlvbnMuYXV0by5jVkk5NmhEaEVlYXd2QW9raVNGQ3N3JnV0bV9tZWRpdW09ZW1haWwiLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Y2RiN2IxMjYxM2I3NzMzMjAzZWU0YzQ3NzNhNDYxZTUyOThjNzc4OWE5MmNmMDAwMDAwMDExMjk4YTY1YzkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1ZJOTZoRGhFZWF3dkFva2lTRkNzdyJ9fSwidXNlcklkIjotMX0.qztPqdQ1FGANU3Yn7AEVCqX8eM0tIY5oTNLmYLJPDt0> 
  |  Android
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3BsYXkuZ29vZ2xlLmNvbS9zdG9yZS9hcHBzL2RldGFpbHM_aWQ9b3JnLmNvdXJzZXJhLmFuZHJvaWQmdXRtX21lZGl1bT1lbWFpbCZ1dG1fc291cmNlPWludml0YXRpb25zJnV0bV9jYW1wYWlnbj1ub3RpZmljYXRpb25zLmF1dG8uY1ZJOTZoRGhFZWF3dkFva2lTRkNzdyIsInRyYWNraW5nIjp7InVzZXJJZCI6LTEsInVzZXJFbWFpbCI6InJlcGx5KzAwOGQzMjRjZGI3YjEyNjEzYjc3MzMyMDNlZTRjNDc3M2E0NjFlNTI5OGM3Nzg5YTkyY2YwMDAwMDAwMTEyOThhNjVjOTJhMTY5Y2UwNzY0Nzg3NEByZXBseS5naXRodWIuY29tIiwibm90aWZpY2F0aW9uVHlwZSI6Imdyb3d0aC5pbnZpdGF0aW9uIiwiY2FtcGFpZ25JZCI6Im5vdGlmaWNhdGlvbnMuYXV0by5jVkk5NmhEaEVlYXd2QW9raVNGQ3N3In19LCJ1c2VySWQiOi0xfQ.72Y4jGg9cuwnx79FTgdIbm_keDdbMx2JgirM8H0Wly0>
Learner Help Center
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL2xlYXJuZXIuY291cnNlcmEuaGVscC8_dXRtX21lZGl1bT1lbWFpbCZ1dG1fc291cmNlPWludml0YXRpb25zJnV0bV9jYW1wYWlnbj1ub3RpZmljYXRpb25zLmF1dG8uY1ZJOTZoRGhFZWF3dkFva2lTRkNzdyIsIn [... cut ...] NWSTk2aERoRWVhd3ZBb2tpU0ZDc3ciLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Y2RiN2IxMjYxM2I3NzMzMjAzZWU0YzQ3NzNhNDYxZTUyOThjNzc4OWE5MmNmMDAwMDAwMDExMjk4YTY1YzkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1ZJOTZoRGhFZWF3dkFva2lTRkNzdyJ9fSwidXNlcklkIjotMX0.a_bRbizCvxNbePhezVN5iZYt7luhRVgncMFcRcznu4c>
Copyright © 2016, Coursera Inc. | 381 E. Evelyn Ave, Mountain View, CA 
94041 USA

WBB-Badtaste, 2016-05-03:
Take the world's best courses, online. Coursera envisions a world where 
anyone, anywhere can transform their life by accessing the world’s best 
learning experience.
[Ready to learn something new?]
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3d3dy5jb3Vyc2VyYS5vcmcvP3V0bV9tZWRpdW09ZW1haWwmdXRtX3NvdXJjZT1pbnZpdGF0aW9ucyZ1dG1fY2FtcGFpZ249bm90aWZpY2F0aW9ucy5hdXRvLmNVcE8wUkRoRWVhOFV4SkVUNWlzY3ciLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Y2I2MGJjMGYyMTU2MjlhYTczMDNkNGNkZjY4YmMxYmYzMmY5MmJjZjU5MmNmMDAwMDAwMDExMmFlZTE4MjkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1VwTzBSRGhFZWE4VXhKRVQ1aXNjdyJ9fSwidXNlcklkIjotMX0.D1pAPnnqxeVxqGraLaJd0gb8bbAWsyFt41AH4S5Yu-U> 	
My Courses
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3d3dy5jb3Vyc2VyYS5vcmcvP3V0bV9tZWRpdW09ZW1haWwmdXRtX3NvdXJjZT1pbnZpdGF0aW9ucyZ1dG1fY2FtcGFpZ249bm90aWZpY2F0aW9ucy5hdXRvLmNVcE8wUkRoRWVhOFV4SkVUNWlzY3ciLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Y2I2MGJjMGYyMTU2MjlhYTczMDNkNGNkZjY4YmMxYmYzMmY5MmJjZjU5MmNmMDAwMDAwMDExMmFlZTE4MjkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1VwTzBSRGhFZWE4VXhKRVQ1aXNjdyJ9fSwidXNlcklkIjotMX0.D1pAPnnqxeVxqGraLaJd0gb8bbAWsyFt41AH4S5Yu-U> 
Course Catalog
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3d3dy5jb3Vyc2VyYS5vcmcvY291cnNlcz91dG1fbWVkaXVtPWVtYWlsJnV0bV9zb3VyY2U9aW52aXRhdGlvbnMmdXRtX2NhbXBhaWduPW5vdGlmaWNhdGlvbnMuYXV0by5jVXBPMFJEaEVlYThVeEpFVDVpc2N3IiwidHJhY2tpbmciOnsidXNlcklkIjotMSwidXNlckVtYWlsIjoicmVwbHkrMDA4ZDMyNGNiNjBiYzBmMjE1NjI5YWE3MzAzZDRjZGY2OGJjMWJmMzJmOTJiY2Y1OTJjZjAwMDAwMDAxMTJhZWUxODI5MmExNjljZTA3NjQ3ODc0QHJlcGx5LmdpdGh1Yi5jb20iLCJub3RpZmljYXRpb25UeXBlIjoiZ3Jvd3RoLmludml0YXRpb24iLCJjYW1wYWlnbklkIjoibm90aWZpY2F0aW9ucy5hdXRvLmNVcE8wUkRoRWVhOFV4SkVUNWlzY3cifX0sInVzZXJJZCI6LTF9.h39KtnmgEfOj4ckxuEBKlTgz8mhZSx9upXo8d9SodNA>

吴彬彬 thinks you'll love learning on Coursera

We envision a world where anyone, anywhere can transform their life by 
accessing the world’s best learning experience. We offer hundreds open 
online courses, in dozens of subject areas, all from top universities and 
institutions.

Course topics range from business and management, to computer science, to 
medicine, music, education, and more.

Millions of our learners have advanced their careers, discovered new 
passions, and built lasting friendships through their Coursera courses. We 
hope you'll join us in learning without limits!

The Coursera Team

Explore Coursera
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3d3dy5jb3Vyc2VyYS5vcmcvb25ib2FyZGluZy8_dG9rZW49OWMxOGFhMTFkOGY5MzJkMjMxYzE5NjY3NWI5OGZiNWUmZW1haWw9cmVwbHkrMDA4ZDMyNGNiNjBiYzBmMjE1NjI5YWE3MzAzZDRjZGY2OGJjMWJmMzJmOTJiY2Y1OTJjZjAwMDAwMDAxMTJhZWUxODI5MmExNjljZTA3NjQ3ODc0QHJlcGx5LmdpdGh1Yi5jb20mbmFtZT10cGF2aW90L3B5dGhvbm9jYy1jb3JlP3V0bV9tZWRpdW09ZW1haWwmdXRtX3NvdXJjZT1pbnZpdGF0aW9ucyZ1dG1fY2FtcGFpZ249bm90aWZpY2F0aW9ucy5hdXRvLmNVcE8wUkRoRWVhOFV4SkVUNWlzY3ciLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Y2I2MGJjMGYyMTU2MjlhYTczMDNkNGNkZjY4YmMxYmYzMmY5MmJjZjU5MmNmMDAwMDAwMDExMmFlZTE4MjkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1VwTzBSRGhFZWE4VXhKRVQ1aXNjdyJ9fSwidXNlcklkIjotMX0.Uc9_0nH6SHiLgW_0epBvSrx6M-SE9l5_yPwDINCMkek>
Download our App:  iOS
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL2l0dW5lcy5hcHBsZS5jb20vYXBwL2FwcGxlLXN0b3JlL2lkNzM2NTM1OTYxP2N0PU1vYmlsZSUyMFByb21vJTIwRW1haWwmdXRtX3NvdXJjZT1pbnZpdGF0aW9ucyZtdD04JnB0PTIzMzQxNTAmdXRtX2NhbXBhaWduPW5vdGlmaWNhdGlvbnMuYXV0by5jVXBPMFJEaEVlYThVeEpFVDVpc2N3JnV0bV9tZWRpdW09ZW1haWwiLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Y2I2MGJjMGYyMTU2MjlhYTczMDNkNGNkZjY4YmMxYmYzMmY5MmJjZjU5MmNmMDAwMDAwMDExMmFlZTE4MjkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1VwTzBSRGhFZWE4VXhKRVQ1aXNjdyJ9fSwidXNlcklkIjotMX0.zbEhRZdE0aY1UZ7oYPso4eZfPZIchFK8HACV--27iDc> 
  |  Android
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3BsYXkuZ29vZ2xlLmNvbS9zdG9yZS9hcHBzL2RldGFpbHM_aWQ9b3JnLmNvdXJzZXJhLmFuZHJvaWQmdXRtX21lZGl1bT1lbWFpbCZ1dG1fc291cmNlPWludml0YXRpb25zJnV0bV9jYW1wYWlnbj1ub3RpZmljYXRpb25zLmF1dG8uY1VwTzBSRGhFZWE4VXhKRVQ1aXNjdyIsInRyYWNraW5nIjp7InVzZXJJZCI6LTEsInVzZXJFbWFpbCI6InJlcGx5KzAwOGQzMjRjYjYwYmMwZjIxNTYyOWFhNzMwM2Q0Y2RmNjhiYzFiZjMyZjkyYmNmNTkyY2YwMDAwMDAwMTEyYWVlMTgyOTJhMTY5Y2UwNzY0Nzg3NEByZXBseS5naXRodWIuY29tIiwibm90aWZpY2F0aW9uVHlwZSI6Imdyb3d0aC5pbnZpdGF0aW9uIiwiY2FtcGFpZ25JZCI6Im5vdGlmaWNhdGlvbnMuYXV0by5jVXBPMFJEaEVlYThVeEpFVDVpc2N3In19LCJ1c2VySWQiOi0xfQ._IuZNbEPRQu9PCb3OhgB-ObxM7WrH16qnXnsCd-pUPY>
Learner Help Center
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL2xlYXJuZXIuY291cnNlcmEuaGVscC8_dXRtX21lZGl1bT1lbWFpbCZ1dG1fc291cmNlPWludml0YXRpb25zJnV0bV9jYW1wYWlnbj1ub3RpZmljYXRpb25zLmF1dG8uY1VwTzBSRGhFZWE4VXhKRVQ1aXNjdyIsIn [... cut ...] NVcE8wUkRoRWVhOFV4SkVUNWlzY3ciLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Y2I2MGJjMGYyMTU2MjlhYTczMDNkNGNkZjY4YmMxYmYzMmY5MmJjZjU5MmNmMDAwMDAwMDExMmFlZTE4MjkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1VwTzBSRGhFZWE4VXhKRVQ1aXNjdyJ9fSwidXNlcklkIjotMX0.3NCNbxiluONIL_GPLexplSr0IItGTqp38SXQc49aR_4>
Copyright © 2016, Coursera Inc. | 381 E. Evelyn Ave, Mountain View, CA 
94041 USA

WBB-Badtaste, 2016-05-03:
Take the world's best courses, online. Coursera envisions a world where 
anyone, anywhere can transform their life by accessing the world’s best 
learning experience.
[Ready to learn something new?]
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3d3dy5jb3Vyc2VyYS5vcmcvP3V0bV9tZWRpdW09ZW1haWwmdXRtX3NvdXJjZT1pbnZpdGF0aW9ucyZ1dG1fY2FtcGFpZ249bm90aWZpY2F0aW9ucy5hdXRvLmNWSVcyUkRoRWVhd3ZBb2tpU0ZDc3ciLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Yzk3MGQyMTE3OTVjMjQ4M2IwZjdjMDgyMDE2MDMxNTA2OGM1Mzc2Zjc5MmNmMDAwMDAwMDExMjk4YmFlZjkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1ZJVzJSRGhFZWF3dkFva2lTRkNzdyJ9fSwidXNlcklkIjotMX0.koUmr7lFMe0AJZ1m_IYCh0-QBcVwFppoKEyx-VNVoX4> 	
My Courses
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3d3dy5jb3Vyc2VyYS5vcmcvP3V0bV9tZWRpdW09ZW1haWwmdXRtX3NvdXJjZT1pbnZpdGF0aW9ucyZ1dG1fY2FtcGFpZ249bm90aWZpY2F0aW9ucy5hdXRvLmNWSVcyUkRoRWVhd3ZBb2tpU0ZDc3ciLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Yzk3MGQyMTE3OTVjMjQ4M2IwZjdjMDgyMDE2MDMxNTA2OGM1Mzc2Zjc5MmNmMDAwMDAwMDExMjk4YmFlZjkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1ZJVzJSRGhFZWF3dkFva2lTRkNzdyJ9fSwidXNlcklkIjotMX0.koUmr7lFMe0AJZ1m_IYCh0-QBcVwFppoKEyx-VNVoX4> 
Course Catalog
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3d3dy5jb3Vyc2VyYS5vcmcvY291cnNlcz91dG1fbWVkaXVtPWVtYWlsJnV0bV9zb3VyY2U9aW52aXRhdGlvbnMmdXRtX2NhbXBhaWduPW5vdGlmaWNhdGlvbnMuYXV0by5jVklXMlJEaEVlYXd2QW9raVNGQ3N3IiwidHJhY2tpbmciOnsidXNlcklkIjotMSwidXNlckVtYWlsIjoicmVwbHkrMDA4ZDMyNGM5NzBkMjExNzk1YzI0ODNiMGY3YzA4MjAxNjAzMTUwNjhjNTM3NmY3OTJjZjAwMDAwMDAxMTI5OGJhZWY5MmExNjljZTA3NjQ3ODc0QHJlcGx5LmdpdGh1Yi5jb20iLCJub3RpZmljYXRpb25UeXBlIjoiZ3Jvd3RoLmludml0YXRpb24iLCJjYW1wYWlnbklkIjoibm90aWZpY2F0aW9ucy5hdXRvLmNWSVcyUkRoRWVhd3ZBb2tpU0ZDc3cifX0sInVzZXJJZCI6LTF9.iBEORQfSIJ5E2f6T1OAE33CeEiPqgVTsvRV206k1B0Q>

吴彬彬 thinks you'll love learning on Coursera

We envision a world where anyone, anywhere can transform their life by 
accessing the world’s best learning experience. We offer hundreds open 
online courses, in dozens of subject areas, all from top universities and 
institutions.

Course topics range from business and management, to computer science, to 
medicine, music, education, and more.

Millions of our learners have advanced their careers, discovered new 
passions, and built lasting friendships through their Coursera courses. We 
hope you'll join us in learning without limits!

The Coursera Team

Explore Coursera
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3d3dy5jb3Vyc2VyYS5vcmcvb25ib2FyZGluZy8_dG9rZW49ZGMxMmI1ODUwOTI2MzU5MzE5YzkwMjA5ZDgyMGMwOTImZW1haWw9cmVwbHkrMDA4ZDMyNGM5NzBkMjExNzk1YzI0ODNiMGY3YzA4MjAxNjAzMTUwNjhjNTM3NmY3OTJjZjAwMDAwMDAxMTI5OGJhZWY5MmExNjljZTA3NjQ3ODc0QHJlcGx5LmdpdGh1Yi5jb20mbmFtZT10cGF2aW90L3B5dGhvbm9jYy1jb3JlP3V0bV9tZWRpdW09ZW1haWwmdXRtX3NvdXJjZT1pbnZpdGF0aW9ucyZ1dG1fY2FtcGFpZ249bm90aWZpY2F0aW9ucy5hdXRvLmNWSVcyUkRoRWVhd3ZBb2tpU0ZDc3ciLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Yzk3MGQyMTE3OTVjMjQ4M2IwZjdjMDgyMDE2MDMxNTA2OGM1Mzc2Zjc5MmNmMDAwMDAwMDExMjk4YmFlZjkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1ZJVzJSRGhFZWF3dkFva2lTRkNzdyJ9fSwidXNlcklkIjotMX0.TW6OAcR8a-wz-dwrBs3Y8Ro98eSRoTZnawhj82d1kWc>
Download our App:  iOS
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL2l0dW5lcy5hcHBsZS5jb20vYXBwL2FwcGxlLXN0b3JlL2lkNzM2NTM1OTYxP2N0PU1vYmlsZSUyMFByb21vJTIwRW1haWwmdXRtX3NvdXJjZT1pbnZpdGF0aW9ucyZtdD04JnB0PTIzMzQxNTAmdXRtX2NhbXBhaWduPW5vdGlmaWNhdGlvbnMuYXV0by5jVklXMlJEaEVlYXd2QW9raVNGQ3N3JnV0bV9tZWRpdW09ZW1haWwiLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Yzk3MGQyMTE3OTVjMjQ4M2IwZjdjMDgyMDE2MDMxNTA2OGM1Mzc2Zjc5MmNmMDAwMDAwMDExMjk4YmFlZjkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1ZJVzJSRGhFZWF3dkFva2lTRkNzdyJ9fSwidXNlcklkIjotMX0.V9t1uNtFkPLsX_brOIvYEyo5xzzBGyglgTlPXQVVR-0> 
  |  Android
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL3BsYXkuZ29vZ2xlLmNvbS9zdG9yZS9hcHBzL2RldGFpbHM_aWQ9b3JnLmNvdXJzZXJhLmFuZHJvaWQmdXRtX21lZGl1bT1lbWFpbCZ1dG1fc291cmNlPWludml0YXRpb25zJnV0bV9jYW1wYWlnbj1ub3RpZmljYXRpb25zLmF1dG8uY1ZJVzJSRGhFZWF3dkFva2lTRkNzdyIsInRyYWNraW5nIjp7InVzZXJJZCI6LTEsInVzZXJFbWFpbCI6InJlcGx5KzAwOGQzMjRjOTcwZDIxMTc5NWMyNDgzYjBmN2MwODIwMTYwMzE1MDY4YzUzNzZmNzkyY2YwMDAwMDAwMTEyOThiYWVmOTJhMTY5Y2UwNzY0Nzg3NEByZXBseS5naXRodWIuY29tIiwibm90aWZpY2F0aW9uVHlwZSI6Imdyb3d0aC5pbnZpdGF0aW9uIiwiY2FtcGFpZ25JZCI6Im5vdGlmaWNhdGlvbnMuYXV0by5jVklXMlJEaEVlYXd2QW9raVNGQ3N3In19LCJ1c2VySWQiOi0xfQ.xkle3DM2VKKIQL1R622EKodlzwMQg07c0H-_EQZKMaw>
Learner Help Center
<https://eventing.coursera.org/redirectSigned/eyJrZXkiOiJlbWFpbC5saW5rLm9wZW4iLCJ2YWx1ZSI6eyJ1cmwiOiJodHRwczovL2xlYXJuZXIuY291cnNlcmEuaGVscC8_dXRtX21lZGl1bT1lbWFpbCZ1dG1fc291cmNlPWludml0YXRpb25zJnV0bV9jYW1wYWlnbj1ub3RpZmljYXRpb25zLmF1dG8uY1ZJVzJSRGhFZWF3dkFva2lTRkNzdyIsIn [... cut ...] NWSVcyUkRoRWVhd3ZBb2tpU0ZDc3ciLCJ0cmFja2luZyI6eyJ1c2VySWQiOi0xLCJ1c2VyRW1haWwiOiJyZXBseSswMDhkMzI0Yzk3MGQyMTE3OTVjMjQ4M2IwZjdjMDgyMDE2MDMxNTA2OGM1Mzc2Zjc5MmNmMDAwMDAwMDExMjk4YmFlZjkyYTE2OWNlMDc2NDc4NzRAcmVwbHkuZ2l0aHViLmNvbSIsIm5vdGlmaWNhdGlvblR5cGUiOiJncm93dGguaW52aXRhdGlvbiIsImNhbXBhaWduSWQiOiJub3RpZmljYXRpb25zLmF1dG8uY1ZJVzJSRGhFZWF3dkFva2lTRkNzdyJ9fSwidXNlcklkIjotMX0.5NHCCRpkJ-gzVHHT29uvQdlodS3SwiIGl2m4ts7DqmI>
Copyright © 2016, Coursera Inc. | 381 E. Evelyn Ave, Mountain View, CA 
94041 USA