tpaviot/pythonocc-core #183: volume maker example

jf---, 2016-01-06:
see [this post](http://dev.opencascade.org/index.php?q=node/1062) provides an important API for splitting. requires OCE 0.17

jf---, 2016-04-08:
hi @tpaviot 

So I'm trying to figure out in python-generator why ``BOPAlgo_MakerVolume`` is not wrapped.
The module looks straightfwd enough ( and compiles ), though the method wont show up in the generated bindings... 

What's your take on this?

python-generator
```python
            'TKBO': ['IntTools', 'BRepAlgoAPI', 'BOPCol', 'BOPInt', 'BOPDS',
                     'BOPAlgo', 'BOPTools'],
```

pyocc session
```python
In [1]: from OCC import BOPAlgo

In [2]: BOPAlgo.
BOPAlgo.BOPAlgo_BadType                  BOPAlgo.BOPAlgo_NonRecoverableFace
BOPAlgo.BOPAlgo_COMMON                   BOPAlgo.BOPAlgo_NotValid
BOPAlgo.BOPAlgo_CUT                      BOPAlgo.BOPAlgo_OperationAborted
BOPAlgo.BOPAlgo_CUT21                    BOPAlgo.BOPAlgo_SECTION
BOPAlgo.BOPAlgo_CheckUnknown             BOPAlgo.BOPAlgo_SelfIntersect
BOPAlgo.BOPAlgo_FUSE                     BOPAlgo.BOPAlgo_TooSmallEdge
BOPAlgo.BOPAlgo_GeomAbs_C0               BOPAlgo.BOPAlgo_UNKNOWN
BOPAlgo.BOPAlgo_IncompatibilityOfEdge    BOPAlgo.SwigPyIterator
BOPAlgo.BOPAlgo_IncompatibilityOfFace    BOPAlgo.SwigPyIterator_swigregister
BOPAlgo.BOPAlgo_IncompatibilityOfVertex  BOPAlgo.new_instancemethod
BOPAlgo.BOPAlgo_InvalidCurveOnSurface    BOPAlgo.register_handle
```

tpaviot, 2016-04-11:
Fixed by commit e1798531de10ffda4ca8f6c44259e0e9c2919a01

closed 2016-04-11 by tpaviot

jf---, 2016-04-11:
@tpaviot awesome! I'll work out an example, cool... :tulip: