tpaviot/pythonocc-core #185: numpy interface for tesselator

jf---, 2016-01-07:
get the mesh and edge data as numpy arrays

rainman110, 2016-01-07:
As cool as this would be, I think it would not be worth to add new dependencies for building pythonocc-core. Also, if have no clue about the C interface of numpy.

jf---, 2016-01-08:
:-1: dealing with mesh data with lists / for loops is not a realistic foresight...
if you really want to built another rendering backend, this is the way to go...

also, numpy is one of python's most pervasive modules, so I think that's ok...
finally, the API I have in mind would look like:

-> find the number of faces of the current mesh
-> create a numpy array of that size
-> use this array as a by-reference argument to a method for the Tesselator class, which will fill it with vertex / face || vertex edge data

now, this way numpy is only optionally required when building the module...
if numpy is not found, just dont built those methods when compiling

if you have numpy fine, use that specialized method, if you dont, use the current API...

does that seem a reasonable way to go about it @rainman110 ?

in the previous pythonocc repo we ( uh, I ) werent taking the issue of dependencies too much into consideration, which was a disaster. your comment is not lost on me :sunglasses:

rainman110, 2016-01-08:
@jf---  Okay, we could make the compilation optional. This way, we could handle the dependencies in a better way.

I have another idea how we could integrate numpy with the tesselator. Instead of polluting the Tesselator class with a new dependency, we could create another class (lets say NumpyTesselator), which uses the Tesselator under the hood. It would copy the values from Tesselator into numpy arrays internally. The copy overhead should be negligible.

We could implement this NumpyTesselator as a cython module, such that we don't have to fiddle around with the python / numpy api. The cython-generated code could be shipped with pythonocc-core.

We still have to think about, in which way the mesh is stored in numpy arrays (i.e. the api of the tesselator), as we could end up looping over the numpy entries again when trying to use the mesh data.

jf---, 2016-01-08:
@rainman110 , lets do it that way...

+1 for doing this the cython way, that makes for a trivial implementation... that's way better :+1:

tpaviot, 2016-01-09:
-1 for using cython or adding any dependency to numpy at the C level, even if optional.

jf---, 2016-01-09:
Ah, I see mixing SWIG & cython is perhaps messy

tpaviot, 2016-01-10:
yes, messy absolutely. However I think it can be achieved at the python level using the numpy frombuffer method.

IMO the best solution would be to:

* modify as you want the Tesselator.cpp class *but* without adding any dependency to another library (I mean something else that needs to be installed in order to get pythonocc compile properly)

and/or

* modify as you want the Visualization.i SWIG file in order to add a numpy export feature *but* don't add any new dependency to numpy or something else

and/or

* do whatever you want at the python level. You can add a conditionnal ```ìmport numpy``` but take care that is properly handled on any machine since numpy must not become a required depency of pythonocc

rainman110, 2016-01-10:
We don't need to mix cython and swig. We can use cython, working on the c++ header file of the Tesselator instead of using its python module.

This has following benefits:
   - everything is done at the python level
   - no c dependency to numpy or any other lib
   - very easy to implement

I see no other way to do it with the restrictions that @tpaviot proposed.

jf---, 2016-01-10:
@rainman110 , fair to point out that the result of cython is plain vanilla C code... that's an interesting thought...

jf---, 2016-04-08:
@rainman110 , thinking about the design of these methods;

- would it be possible to create the numpy arrays prior to setting the data in python?  
  - this way we could have the numpy arrays passed by-ref in the Tesselator method, and sidestep the issue of having a numpy dependency when building pythonocc...

curious to have your thoughts ;)

jf---, 2016-09-28:
Gents, I'm still very curious about this, since it would open a lot of possibilities in terms of OpenGL backend ( qml 3d? raw opengl? qt? ). IMO it would be best to go ahead with a cython version, discuss whether its acceptable ( true that its "just" C code ) and take it from there?

I've done some work with assimp that lets you load meshes blazingly fast... just pushing those vertices to OpenGL is what takes long because of the loops.

Same for more advanced OpenGL rendering, for that the cython approach is perfect.
For raw speed, cython is just amazing, so also in this regard ( sidestepping bottlenecks ) I think its interesting.

I can see from @tpaviot that this might be better for **pythonocc-utils** rather then **pythonocc-core**

jf---, 2016-09-29:
Gents, so here's a [reference](http://stackoverflow.com/questions/28904033/extracting-swig-wrapped-c-instance-pointer-for-use-in-cython) on how to get a pointer from SWIG and access it in cython. This way, the OpenGL from pyocc can be accessed.

jf---, 2016-10-29:
[swig example](https://gist.github.com/mattbierbaum/1189856) of looping a numpy array

tpaviot, 2018-01-25:
Much simpler to return python lists from SWIG and perform a numpy conversion at the python level. Implemented in the jupter_renderer. Closing this issue.

closed 2018-01-25 by tpaviot