tpaviot/pythonocc-core #187: integrate osx platform in travis

jf---, 2016-01-08:


tpaviot, 2016-01-09:
Work started at branch review/travis-osx.

Seems that only a few changes a required over linux config, since osx can use conda exactly the same way on both platforms (see 9a16387b823d4841f6aa3197b239d95a35a07ccf). This is quite useful rather than using apt on Linux and brew on osx.

However I did not manage to get the compilation successfull. It failts at linking against oce, see for instance:

https://travis-ci.org/tpaviot/pythonocc-core/jobs/101227267

tpaviot, 2016-01-10:
There's an issue with the binary provided with conda for osx-64. On travis, all the methods that take a stream as a parameter fails to link to oce libraries (see for instance https://travis-ci.org/tpaviot/pythonocc-core/jobs/101227267). @jf--- @rainman110 your opinion about that ? where does the dlr-sc binary come from (os version, clang version ) ?

tpaviot, 2016-01-10:
it seems ti has to do with stdc++ linking see http://stackoverflow.com/questions/12920891/std-linker-error-with-apple-llvm-4-1

rainman110, 2016-01-10:
It has something to do with the deployment target on OS X. Conda packages are beeing build with target 10.6:

    http://conda.pydata.org/docs/building/environment-vars.html 

Therefore, you must use the same for building pythonocc-core against oce. Just use

     export MACOSX_DEPLOYMENT_TARGET=10.6

before the build. That should resolve the issue.

rainman110, 2016-01-10:
Now, the next problem occured. Hardcoded system paths like in some previous issue (which I can't find anymore). I am gonna fix these paths in conda/oce.

rainman110, 2016-01-10:
I removed the system paths from the conda/oce package. I've rebuild oce. Now, pythonocc should build fine.

tpaviot, 2016-01-10:
Thanks, that worked, see https://travis-ci.org/tpaviot/pythonocc-core/jobs/101450776

Still other issues to fix though.

rainman110, 2016-01-10:
This problem looks familiar. I had it once on OS X. 

You get the error, when two different python.dylib are loading. This can happen, if the RPATH is not properly set up. In these cases, python(exe) loads the system python.dylib, whereas pythonocc-core loads the conda python.dylib. 

Please set the environment variable DYLD_PRINT_LIBRARIES to allow debugging this issue.

tpaviot, 2016-01-11:
@rainman110 ok, see https://travis-ci.org/tpaviot/pythonocc-core/jobs/101518860

tpaviot, 2016-01-11:
@rainman it seems you're right, see https://travis-ci.org/tpaviot/pythonocc-core/jobs/101518859#L1624

_Standard.so loads /usr/lib/libpython2.7.dylib whereas /Users/travis/miniconda/bin/../lib/libpython2.7.dylib is already loaded

rainman110, 2016-01-11:
@tpaviot  Interesting. So let my explain, what happens: Python loads the miniconda python.dylib. Pythonocc-core loads the system python.dylib.

I can think of two ways to fix the build:
  1. Remove the system python
  2. Or, fix the rpath of the pythonocc-module to search for the miniconda python first. This is actually done by conda build to make the conda package relocatable. You can use the tool install_name_tool from OS X to achieve this.

Can you please use ```otool -l``` to extract the lib dependencies and search paths for one of the pythonocc modules?

rainman110, 2016-01-11:
By the way, a more safe way for Continous Delivery with Conda packages would be, to actaally use conda to build pythonocc-core. This way you'll get a relocatable package that you can upload directly.

To achieve this, you just need a conda recipe inside the pythonocc-core source.

tpaviot, 2016-01-11:
@rainman110 Agree with you to include the conda receipe into pythonocc-core to be able to upload conda binaries. However, I'd like first to experiment building on osx.

tpaviot, 2016-01-11:
@rainman110 removing the system python lib does not work. https://travis-ci.org/tpaviot/pythonocc-core/jobs/101530456

rainman110, 2016-01-11:
@tpaviot  Here's the problem:

https://travis-ci.org/tpaviot/pythonocc-core/jobs/101530456#L1919

libpython2.7.dylib has no search path. Instead, it is solely referenced by its name.

The entry should be changed from libpython2.7.dylib to @rpath/libpython2.7.dylib.

This can be done be executing:
    
    install_name_tool -change _Standard.so libpython2.7.dylib @rpath/libpython2.7.dylib

This has to be done for all pythonocc module obviously.

jf---, 2016-01-11:
@rainman110 , conda handles this for you automagically, with the __binary_relocation__ option
( however, in certain cases re-writing the libs takes up quite a lot of time )

```yaml
build:
  number: 1
  binary_relocation: true

```

rainman110, 2016-01-11:
@jf--- I know. Binary relocation is also enabled default. In this case here, @tpaviot  wants to build it without conda. Therefore he has to do it manually.

Also, the binary relocation of pythonocc-core made by conda-build is taking really long (more than 1.5 hours) on my macbook pro. This is much more time than the build process itself. Does it work faster on your machine?

jf---, 2016-01-11:
Depends: remember our discussion on @rpath / @loader_path ?
I think conda wont rewrite paths with @rpath...

This is something to look into, yes, re-writing the paths is taking longer than the compilation ;(
So in terms of performance, its a delicate issue

rainman110, 2016-01-11:
@jf--- Yes, i remember the discussion :smirk:  . Since then, I read a lot about this stuff. Conda does a good job in fixing all these loader path things (including __@rpath__) by default. It actually does not change many entries though, as most of them are already correct. The only entry that has to be fixed is the python path. 

For building pythonocc-core without conda, these paths have to be fixed as described above. For the build without conda, nothing has to be done (only if we want to speed up the build)

jf---, 2016-01-11:
Great that you've learned more on this topics ( are we doing "the right thing" with @rpath ? ) 
:+1: if we can speed things up with a simple python script, the conda approach is just too slow...

rainman110, 2016-01-11:
@jf---  I am actually not sure, what we are currently doing with __@rpath__.? Is there any rpath specific stuff in the repo?

Fact is, that pythonocc-core is beeing built correctly. The python path has to be fixed just because python's install_name has no search path which either conda adds or we could add instead of conda to speed things up.

tpaviot, 2016-01-11:
@rainman110 ok, it looks safer to use conda to build pythonocc. I removed the testing stage so that compilation is fine, see https://travis-ci.org/tpaviot/pythonocc-core/builds/101578931. Build passes.

I think this issue can be closed. In the future, let's just rebase review/travis-osx on master if ever we want the compilation be launched on osx without conda.

rainman110, 2016-01-11:
@tpaviot I am working on this. I'll provide the conda recipe with a PR.

tpaviot, 2016-01-11:
@rainman110 Tryin to build the conda recipe on travis, failing because of a missing ninja (but installed from dlrsc repository) see https://travis-ci.org/tpaviot/pythonocc-core/jobs/101633207

rainman110, 2016-01-11:
@tpaviot Hang on, I am working with it. Ninja is not missing, you have to add the DLR channel.

But just hang on, you'll get a PR in a couple of hours that does everything.

rainman110, 2016-01-11:
Here an intermediate status:

https://travis-ci.org/rainman110/pythonocc-core/jobs/101673046

tpaviot, 2016-01-11:
interesting, I see what I missed

rainman110, 2016-01-11:
Update: https://travis-ci.org/rainman110/pythonocc-core/builds/101682146

Strangely, the gcc compilers are crashing  :(  With clang it works

tpaviot, 2016-01-11:
Here that works, with both gcc and clang https://travis-ci.org/tpaviot/pythonocc-core/builds/101675338

rainman110, 2016-01-11:
I suppose it was using too much memory due to ninja parallelism. I switched back to make. Hopefully this fixes it.

rainman110, 2016-01-11:
@tpaviot Please note, that in your build matrix gcc 4.8 is used for all jobs, even if the CC/CXX variables are set correctly.

rainman110, 2016-01-12:
Hi @tpaviot . I see, that you are struggeling with the CXX / CC compiler vars.

Instead of setting the variables artificially, you can just change the lines
   ```
build:
  script_env:
    - CC
    - CXX
   ```

to 

   ```
build:
  script_env:
    - CC [linux]
    - CXX [linux]
   ```

in the meta.yml. These settings are only required for linux, since you have multiple compilers there. that should solve the problem.

rainman110, 2016-01-12:
@tpaviot  I test build on my travis seem to find the compilers:
https://travis-ci.org/rainman110/pythonocc-core/jobs/101829910

jf---, 2016-01-12:
So much progress... 

@rainman110 , what about running the unittests after the build has completed?
This way, when the tests pass, we could push the build to anaconda.
( sorry if I'm stating the obvious... )

rainman110, 2016-01-12:
Hi @jf---  Yes, of course this should be done. But one step after another :smirk:

jf---, 2016-01-12:
I'll shut up ;)
Awesome to see this come together...

tpaviot, 2016-01-12:
@rainman110 ok, your fix with [linux] only for CC and CXX works.

tpaviot, 2016-01-12:
@rainman110 Regarding the test stage, it could be better to run the unittest rather than only importing 2 pythonocc modules. At the end of the build.sh script, it would be cool to add:
```bash
cd test
python run_tests.py
```
or something like that.

rainman110, 2016-01-12:
Yes of course we have to run the test suite. I had that in mind and wanted to do this. Just struggeling currently with fine details to speed up the build on osx.

rainman110, 2016-01-12:
@jf---  @tpaviot  Has anyone of you experience with ```ccache``` on travis/osx? It would be cool to use ccache to speed up the compilation. The compilation takes about 40 minutes on osx which is almost the deadline when travis jobs get killed.

tpaviot, 2016-01-12:
@rainman110 Never head about ccache before.

tpaviot, 2016-01-13:
Added tests at the end of build.sh. On test doesn't pass (test_bezier_surfaces, see https://travis-ci.org/tpaviot/pythonocc-core/jobs/102024937#L3274). Already reported at #177

jf---, 2016-01-13:
ccache seems very useful... from wiki:

```
ccache is a software development tool that caches the output of C/C++ compilation so that the next time, the same compilation can be avoided and the results can be taken from the cache. This can greatly speed up recompiling time. The detection is done by hashing different kinds of information that should be unique for the compilation and then using the hash sum to identify the cached output. Ccache is licensed under the GNU General Public License.
```

Finally, it seems that the new none-sudo / docker based TravisCI [allows for caching](https://docs.travis-ci.com/user/caching/). So perhaps its possible to us ccache?

jf---, 2016-01-13:
Btw, perhaps I'm getting late to the party, but [Travis' command line client](https://github.com/travis-ci/travis.rb#readme) seems really great, also supporting a console

rainman110, 2016-01-13:
@tpaviot  I suspect a faulty memory managment. The implementation of GetHandle look suspicious to me. Just casting a raw pointer to a Handle can produce akward problems.

If i run core_memory_unitttest.py, I get many errors:
  - Under linux python 2 + 3.4: Two errors (testCollector, testGC)
  - Under windows (python 3.5): Same errors + failure of testNullifyHandle

Did this work in the past?

tpaviot, 2016-01-13:
@rainman110 I didn't work on the Handle part for a while, and core_momety_unittest is not maintained anymore. It used to work, but it surely needs to be rewritten (or removed).

I never really solved the Handle issue in pythonocc. Feel free to modify/suggest anything you want, and open an issue, I'll be happy to share what I've learned so far.

rainman110, 2016-01-13:
Lets make a different issue about the memory management.

rainman110, 2016-01-13:
I have a multi-platform matrix job running. Looks good so far. Of course the tests will still fail on osx.

https://travis-ci.org/rainman110/pythonocc-core/builds/102106474
https://github.com/tpaviot/pythonocc-core/compare/master...rainman110:review/ms_conda_osx

Feel free to integrate this stuff.

closed 2016-01-16 by tpaviot

jf---, 2016-01-16:
hi @tpaviot , what is the plan with the changes to travisci that @rainman110  has proposed?
osx fails because of a failing test, it seems correct otherwise. think its a cool contribution...

tpaviot, 2016-01-16:
osx failure is fixed in another branch

osx stuff can't be merged as is. Mutlisystem feature must be manually set up, I requested travis-ci support to enable this feature

rainman110, 2016-01-18:
@tpaviot What do you mean the multisystem feature must be manually set up. I had a job with os x and linux running successfully.

What about the crashing bezier surface test. Did you figure out the cause of the crash?