tpaviot/pythonocc-core #189: fix appveyor builds for CI / CD

jf---, 2016-01-08:


tpaviot, 2016-01-08:
I've been working on it see branch tp/reveiw/appveyor (https://github.com/tpaviot/pythonocc-core/tree/tp/review/appveyor). Configuration/setup is of for msvc14. The main problem with appveyor is that compilation times can't exceeds 1hr. This is not enough to complete the build. According to these first tests, I doubt appveyor can be used for CI/CD on Windows.

jf---, 2016-01-08:
Is that building first OCE then pythonocc-core Thomas?
That is to say, was this updated like TravisCI to use conda to grab OCE then build pythonocc-core?

Man, that's sad news, I know you have gone through great lengths to make this work...

tpaviot, 2016-01-08:
the appveyor script first downloads install conda and then installs oce binary using conda. This works quite well. Unfortunately pythonocc-core build itakes too much time. For mingw, I didn't succeed yet to configure/setup/build properly, but I'm quite sure it wi be the same. Didn't find any work around or temporary fix yet.

jf---, 2016-01-08:
Thanks for the heads up Thomas...

jf---, 2016-02-17:
@tpaviot, @rainman110 
shall we drop Appveyor support?

jf---, 2016-03-29:
Gents, shall we take a decision on this ( @tpaviot , @rainman110  ).

tpaviot, 2016-03-29:
@jf--- Appveyor is very important, even if the compilation process doesn't go till the end.

rainman110, 2016-03-29:
From my experience, the creation of the cxx files with swig takes most of the time. As this is completely platform independent, one could create these files externally (e.g. checking them into the repository or in some helper repo). This would probably decrease compilation times in much less than one hour.

tpaviot, 2016-08-10:
Good news : Appveyor recently moved to a new architecture (https://www.appveyor.com/blog/2016/07/16/migration-to-rackspace). This drastically improves compilation speed. I successfully experimented a 6x (!) improvement on oce. Hopefully this will also work for pythonocc-core.

tpaviot, 2016-08-18:
I'm currently being working on the https://github.com/tpaviot/pythonocc-core/tree/review/ci-cd branch. It contains major improvements for both appveyor and travis. In a short term, hopefully by mid september, I'll get travis and appveyor build pythonocc/oce for all os/architecutres with an automatic upload to anaconda cloud (already successfully tested with appveyor). pythonocc ditro will be uploaded to https://anaconda.org/pythonocc, oce to https://anaconda.org/pythonocc. @rainman110 dlr-sc repository misses some oce builds (linux 32, vc10 and 14 for oce-0.17.1, all oce-0.17.2 win relases) that makes pythonocc-core build impossible for thoses platforms, all of that will be automated from appveyor/travis.

jf---, 2016-08-18:
@tpaviot , this is exciting news! love the idea that builds will be automagically uploaded... wow...
I did create a [pythonocc](https://anaconda.org/pythonocc) organization on anaconda.
Also, note that multiple persons can collaborate / upload binaries, possible this is practical?

rainman110, 2016-08-18:
These are greate news. Not that I mind creating binaries by hand, but an automated process is much more practical :+1: Also, we at DLR typically want to apply some patches to oce that may not be needed in general. This way, we can decouple the needs of DLR with your own.

rainman110, 2016-08-19:
@tpaviot I looked at your failing appveyor builds. It seems you are struggling with conda.

I also experienced the same error. Since the upgrade to conda 4.x, I get this error whenever I have a package with features (vc9, vc10, vc14) and tests.

The workaround is, to disable conda tests and instead run them directly from appveyor / travis.

This issue was already posted here: https://github.com/conda/conda/issues/1820

It seems, that the conda folks have problems with the dependency resolution of the packages. Many of their bugs come from their solver.

rainman110, 2016-08-19:
I just found out, that appveyor support test publishing.

If our test framework can generate a junit compatible xml output, we can upload it to appveyor:
https://www.appveyor.com/docs/running-tests/#uploading-xml-test-results

rainman110, 2016-10-03:
@tpaviot I would say, this can be closed. Appveyor is working like a charm 😄

jf---, 2016-10-03:
@rainman110 True! thanks for the reminder...

closed 2016-10-03 by jf---