tpaviot/pythonocc-core #190: use verioneer for automated release versioning

jf---, 2016-01-08:
[see python-versioneer](https://github.com/warner/python-versioneer)

tpaviot, 2016-01-08:
On the Versionneer page I can read:

"This is a tool for managing a recorded version number in distutils-based python projects".

pythonocc-core compilation/package is not managed by distutils. @jf--- can u please check it is distutils only and close the issue or leave it open but ensure this is compliant with any distribution system (for instance cmake).

rainman110, 2016-01-08:
Here's a way, to integrate git information into the conda recipe version:
http://conda.pydata.org/docs/building/environment-vars.html#git-environment-variables

This almost suffices. We just need to integrate the git-sha into cmake, which can be done like this:
https://github.com/DLR-SC/tigl/blob/master/cmake/GetGitRevisionDescription.cmake

jf---, 2016-01-08:
Lets keep it simple if we can ;)

closed 2016-01-16 by tpaviot