tpaviot/pythonocc-core #191: use Cont. Integration for Cont. Delivery

jf---, 2016-01-08:
## A commit to master yields an installable conda pythonocc-core package for linux, osx, windows

### TL;DR
* Use CI for CD ( cont. delivery ) -- lets not repeat the effort
* A commit to master is a new release

---

### What about the following outlook:

* a commit to master triggers a travisci build, using conda
* whenever all tests pass, I think we should use __conda upload__ to push the resulting package to the anaconda site
* these builds should go to a separate channel, say, a dev channel
* whenever we want to release a specific package, just change the channel from dev -> main
* then run __conda update pythonocc-core --channel dev__
* this is why [automatic versioning](https://github.com/warner/python-versioneer) is important
  * in addition to Major, Minor, Patch, you'd have another attribute, SHA1, referring to the git commit
  * eg, 0.16.3.6f5280 / 6f52808fba5db219d57dfe8e739ded6e2653c2bc

now, currently, this would cover linux. we can have osx #187 for travis too. 
do the same thing with appveyor for windows

---

## TODO

*[ ] OS X builds with travis #187 
*[ ] Make app-veyor functioning for the windows builds (last time i checked it was not working) #189
*[ ] Use git SHA as version id. We need to add the SHA to the conda recipe and the CMakeLists.txt as well #190

jf---, 2016-01-08:
an anaconda [pythonocc organization](https://anaconda.org/pythonocc/dashboard) is setup. @tpaviot and @rainman110 are already added as project owners

jf---, 2016-01-08:
travisci will require an anaconda account for uploading... added to TODO

tpaviot, 2016-01-08:
An example of howto use travis and appveyor for cd at https://github.com/rmcgibbo/python-appveyor-conda-example

jf---, 2016-01-08:
interesting, in order to upload binaries to anaconda, you need to have key, a token.
it has a good explanation of how this is done securely on appveyor...

very useful...

closed 2016-04-10 by jf---