tpaviot/pythonocc-core #193: DisplayMessage doesnt render text

jf---, 2016-01-08:
minimal example

```python

from OCC.gp import gp_Pnt
from OCC.Display.SimpleGui import init_display
display, start_display, add_menu, add_function_to_menu = init_display()

if __name__ == '__main__':
    P8 = gp_Pnt(-24, 120, 12.5)
    display.DisplayMessage(P1, "Does not show up")
    start_display()

```

jf---, 2016-01-08:
Quite possible that this has something to do with the link issue to freeimage issue experienced installing possible specific to the DLR/SC osx binaries...

I dont believe this to be a pythonocc issue, but build related

jf---, 2016-01-08:
tried with the current DLR/SC build ( 0.16.3 ), same thing...

jf---, 2016-01-08:
better example: pythonocc-core/examples/core_topology_glue.py

tpaviot, 2016-01-09:
works on my windows machine
[image]

jf---, 2016-01-09:
Yep, seems like this is an issue with the binary. I tried both my own built & DLR/SC, same result.

closed 2016-01-09 by jf---

efirvida, 2016-04-18:
I have the same Issue using python2.7 oce-16.1 python-occ 0.16.4 build from source in opensuse leap 42.1

calocedrus, 2017-03-17:
This is closed and apparently not directly related to pythonocc but may I ask how to properly render text (using the nightly built, python 3.5 and linux, I observe the same as what @jf---  reported)? It seems that all convex characters (e.g. 0, a, B, 8,...) will show a strange behviour (characters replaced with vertical lines), but characters made of straight lines are fine (e.g. 1, i, l, k...)

chavan-vjti, 2017-07-10:
Can I get to know, what exactly could be the linking issue with freeimage?