tpaviot/pythonocc-core #194: Special characters in ReadFile(PATH)

FoxP, 2016-01-08:
In the example below (from http://webgl.pythonocc.org/)

```python
from OCC.STEPControl import STEPControl_Reader
from OCC.Display.WebGl import x3dom_renderer
step_reader = STEPControl_Reader()
#PATH = str(askopenfilename(filetypes=[("STEP",("*.stp","*.step"))]))
step_reader.ReadFile(PATH)
step_reader.TransferRoot()
block_cylinder_shape = step_reader.Shape()
my_renderer = x3dom_renderer.X3DomRenderer()
my_renderer.DisplayShape(block_cylinder_shape)
```

If PATH contain special characters, python just crashes.
I have this issue on Python 2.7.6.1 with PythonOCC 0.16.0 on Windows 8.1 64bits.

jf---, 2016-01-08:
Correct, it wont handle non-ascii characters, we should test this before loading.
Thanks...

tpaviot, 2016-01-09:
@FoxP what do you mean with "special character" ? non-ascii ? which enconding to you use ? can you please give an example of a filename that fails to be loaded.

FoxP, 2016-01-09:
A path like this one can crash the example above : "C:\Users\FoxP\Downloads\téléphone.stp"
"é" is a french special character. I'm using utf-8 encoding.

rsm-imap-cc, 2016-01-10:
I tested the problem on GNU/Linux and python3.4
I used the path: PATH='/home/rsm/Desktop/Téléphone.stp' and everything worked fine. I'm using PythonOCC 0.16.2

This may due to the fact that python2 does not automatically converts to Unicode. @FoxP Have you tried after converting to Unicode?

FoxP, 2016-01-11:
With a utf-8 decoded string, i have the following error:
```python
TypeError: in method 'XSControl_Reader_ReadFile', argument 2 of type 'char const *'.
```
```python
path = path.decode('utf-8')
print path
# C:/Users/Paul/Downloads/téléphone.stp
print type(path)
# <type 'unicode'>
step_reader.ReadFile(path)
```

jf---, 2016-01-12:
pythonocc < 0.18 wont support unicode... 
related to the library we're wrapping

FoxP, 2016-01-13:
Thanks for your answer. What encoding should i use so?

jf---, 2016-02-17:
OCE 0.17 supports UTF-8, I'll investigate this...