tpaviot/pythonocc-core #195: IGESControl_Writer crashes python

msaysell, 2016-01-11:
```
from OCC.IGESControl import IGESControl_Writer
writer = IGESControl_Writer()
```

The code above will cause my python (2.7.10) to stop working with the error code `-1073741819`

I'm using at least version 0.16.0 and realise this is a very vague description.... Is there anything at all that would help me diagnose the cause of this crash?

rainman110, 2016-01-11:
I can reproduce it with python 3.5 Windows x64, pythonocc-core 0.16.3. I am looking into this issue.

msaysell, 2016-01-11:
Much appreciated! If you need any further information please let me know! (I'm also using Windows x64)

rainman110, 2016-01-11:
This is an "issue" of opencascade, as the iges controller is not initialized. Please change your code to:

    from OCC.IGESControl import IGESControl_Writer, IGESControl_Controller_Init
    IGESControl_Controller_Init()
    writer = IGESControl_Writer()

Does this fix your problems?

msaysell, 2016-01-11:
Your my hero, yes it does fix my issue, thank you very much!

closed 2016-01-11 by msaysell

jf---, 2016-01-11:
Nice catch @rainman110 !

rainman110, 2016-01-11:
@msaysell You're welcome :smirk: