tpaviot/pythonocc-core #199: Memory management tests fail

rainman110, 2016-01-13:
If i run core_memory_unitttest.py, I get many errors:

    Under linux python 2 + 3.4: Two errors (testCollector, testGC)
    Under windows / python 3.5: Same 2 errors + failure of testNullifyHandle
    Under windows / python 3.4: Same 2 error as under linux

Did this work in the past?

rainman110, 2016-01-13:
I am wondering, how the Handle mechanism works from within python.

What looks strange to me, is that GetHandle just casts a raw pointer.

I don't know the internals, but Handles are reference counted. In C++

    Handle_MyTransient h = new MyTransient()

creates a pointer and a handle, that guards this pointer. Also it increases the refcount of the transient object.

Pythonocc does things like
    
    (Handle_MySelf*) &SelfPointer

which just casts a Transient pointer address to a Handle pointer. Does this suffice to increase the refcount of the Transient object?

@tpaviot Can you please share the ideas behind your handle implementation.

tpaviot, 2016-01-13:
@rainman110 Yes, this implementation is enough to increase reference counting of the transient object. Try this:
```python
$ python
>>> from OCC.Standard import *
>>> MyTransient = Standard_Transient()
>>> MyTransient.GetRefCount()
0
>>> Handle_MyTransient = MyTransient.GetHandle()
>>> MyTransient.GetRefCount()
1
>>> Handle_MyTransient2 = MyTransient.GetHandle()
>>> MyTransient.GetRefCount()
2
```

tpaviot, 2016-01-13:
The Nullify method decreases reference counting:
```
>>> Handle_MyTransient.Nullify()
>>> MyTransient.GetRefCount()
1
```

tpaviot, 2016-01-13:
Tests below are launched on win32/py27

tpaviot, 2016-01-13:
Note that the memory results should depend on the value of the MMGT_OPT en var.

rainman110, 2016-01-13:
My windows / python3.5 test was done on win64. Look at this example:

```python
In [1]: from OCC.Standard import *

In [2]: MyTransient = Standard_Transient()

In [3]: MyTransient.GetRefCount()
Out[3]: 0

In [4]: Handle_MyTransient = MyTransient.GetHandle()

In [5]: MyTransient.GetRefCount()
Out[5]: 1

In [6]: Handle_MyTransient.Nullify()

In [7]: MyTransient.GetRefCount()
Out[7]: 69891209
```

Isn't this strange. Happens only on windows py35. The question is, what happens under the hood.

rainman110, 2016-01-13:
In the swig generated shadow classes, the GarbageCollector module is never imported! Therefore it does not get called. The thrown exception will be ignored by

```python
except:
   pass
```

tpaviot, 2016-01-13:
@rainman110 right

tpaviot, 2016-01-13:
I can't remember where its was imported and when/why it was removed ...

tpaviot, 2016-01-22:
Fixed by PR #200

closed 2016-01-22 by tpaviot