tpaviot/pythonocc-core #204: Make install instructions more accessible

rainman110, 2016-01-21:
Installing pythonocc via conda is easy. However, to find the install instructions is not.

Currently, one has to:
   1. Scroll to the bottom of the github page to __How to compile ?__
   2. Scroll up again to find the Install.md file and click on it
   3. Scroll again down to __Install with conda__

The average user probably does not want to build it by himself, i.e. I would prioritize installing binaries over self building.

Also, I would provide a link on the home page. In principle, there is the "install with conda" badge, but I suspect that anyone notice it in between all the other badges.

tpaviot, 2016-01-22:
@rainman110 I agree.  A post on the ml about howto build/install with conda would be welcome. Indeed, it would be referenced by google and other search engine. And a post on the pythonocc blog too. Feel free tu submit a PR about that

tpaviot, 2016-04-06:
Fixed by commit e70991a9ee81f49e92607aa61ba132a2c95a07d4

closed 2016-04-06 by tpaviot