tpaviot/pythonocc-core #206: oce-0.17.1 support

tpaviot, 2016-01-23:
@rainman110 Is oce-0.17.1 available on ldr-sc repos ?

rainman110, 2016-01-23:
@tpaviot no, not yet. Progressing on pythonocc 0.17 ?

tpaviot, 2016-01-23:
Not yet, but as soon as it's started, we must have travis working, thus oce-0.17.1 conda package available.

I plan to work about the port to oce-0.17 not on the next week, but the week after. I'll work with local tests under Windows/mingw, but travis is really important in this process.

jf---, 2016-02-08:
@tpaviot , as a reminder, there has been a contribution of @colonelzentor
that supports pythonocc-core for OCE 0.17

[See this branch](https://github.com/colonelzentor/pythonocc-generator/commits/vtk-integration)

tpaviot, 2016-02-08:
@jf--- Thanks for the reminder, that's something I absolutely forgot, I can't even remember about this discussion. Hopefully it will save time.

rainman110, 2016-02-09:
Just a note: I am currently creating the conda windows builds for OCE 0.17.

tpaviot, 2016-02-15:
rainman110 I started the port to oce-0.17.1. Let me know as soon as the packages are available on the dlr-sc repo, travis can't build so far because of missing packages for linux/osx.

tpaviot, 2016-02-15:
The branch hosting port to oce-0.17.1 dev is here https://github.com/tpaviot/pythonocc-core/tree/review/tmp-port-to-oce-0.17.1

jf---, 2016-02-15:
Gents; 

I hope its ok to raise 2 questions regarding conda:

- shall we announce this to be the official pythonocc binaries from 0.17+ on?
- right now, the DLR-SC repo is the "official" repo. I've setup a [pythonocc](https://anaconda.org/pythonocc) organization such that we can collaborate on releases.

I think moving to the latter is important before announcing the transition to conda.

@tpaviot, has the prior work porting to 0.17 saved you some hours? By the looks of your commits not much?

rainman110, 2016-02-15:
@tpaviot I've built oce 0.17 for linux. So you could start at least with the linux ci.

When I get home today, I'll create also the osx build.

jf---, 2016-02-15:
I've created a build of OCE 0.17.1 that includes @rainman110 [ Clang fixes ](https://github.com/rainman110/oce/tree/ms/fix-oce-nullrefs). See [here](https://anaconda.org/jf/oce/files). ( though I have more faith in Martin's builds myself :sunglasses:  )

Building review/tmp-port-to-oce-0.17.1 went quite well, but trips over a missing header: **Graphic3d.hxx**

---

Scanning dependencies of target _Visualization
[ 95%] Building CXX object CMakeFiles/_Visualization.dir/Darwin/x86_64-MinSizeRel-64/VisualizationPYTHON_wrap.cxx.o
In file included from /Users/jelleferinga/miniconda/conda-bld/work/Darwin/x86_64-MinSizeRel-64/VisualizationPYTHON_wrap.cxx:3138:
/Users/jelleferinga/miniconda/conda-bld/work/src/Visualization/Visualization.h:28:10: fatal error: 'Graphic3d.hxx'
      file not found
#include <Graphic3d.hxx>
         ^
1 error generated.
make[2]: *** [CMakeFiles/_Visualization.dir/Darwin/x86_64-MinSizeRel-64/VisualizationPYTHON_wrap.cxx.o] Error 1
make[1]: *** [CMakeFiles/_Visualization.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

jf---, 2016-02-15:
**Visualization.h**
---

Remove:

```c++
include <Graphic3d.hxx> for:
```
Change:

```c++
#include <Handle_Graphic3d_GraphicDriver.hxx>
#include <OpenGl_GraphicDriver.hxx>
```
---
**Display3d.cpp**

change:
```c++
  GetGraphicDriver() = Graphic3d::InitGraphicDriver (aDisplayConnection);
```

to:

```c++
  GetGraphicDriver() = new OpenGl_GraphicDriver (Handle(Aspect_DisplayConnection)());
```

---

the [release notes of 6.8.0](http://www.opencascade.com/sites/default/files/documents/Release_Notes_6.8.0.pdf) might come in handy...

there's still quite a nr of issues in Tesselator.cpp

rainman110, 2016-02-15:
The OS X binaries for one 0.17 are on the DLR-SC repo:
https://anaconda.org/DLR-SC/oce/files

@tpaviot You should be able to start on CI for the upcoming pythonocc.

jf---, 2016-02-15:
nice work @rainman110 

---

regarding Tesselator.cpp -- a lot of changes have been made to the BRepMesh package.

@tpaviot , looks like BRepMesh::Mesh -> BRepMesh_IncrementalMesh?

rainman110, 2016-02-15:
@jf--- Yes, this is exactly the case:
BRepMesh::Mesh -> BRepMesh_IncrementalMesh

These are the changes the we had to do in our software. Most of it is BrepMesh. We also have some changes in the visualization as well:

jf---, 2016-02-16:
thanks @rainman110 , that's a very helpful reference...

tpaviot, 2016-02-16:
Thanks @rainman110 and @jf--- , the compilation is succesfull on linux, osx, and on my windows machine. There's still an issue with GUI tests on Linux. On windows it works. @jf--- could you please test on your osx machine that the core_helloworld example pass ?

jf---, 2016-02-16:
hi @tpaviot , all seems well, however when I build using pythonocc-core's conda script, I have an issue linking with libgl2ps that isnt easily fixed with **install_name_tool**...

that said, I'm happy to report:

```bash
py /Users/jelleferinga/GIT/pythonocc-core/test/core_geometry_unittest.py
............Q1: at Distance :4.079132288244904
Q2: at Distance :7.833305801192616
....
----------------------------------------------------------------------
Ran 16 tests in 0.152s
```

tpaviot, 2016-02-16:
@rainman110 IS your oce-0.17.1 build against vtk ?

tpaviot, 2016-02-16:
@jf--- What's the workflow to push to the pythonocc conda repos you created ? I'll experiment with linux 32 bits, currently missing from dlr-sc repos

jf---, 2016-02-16:
Hi @tpaviot ,

Cool, i think it would be great if we can contribute to a shared pythonocc repo.
You'll find the documentation on how to do [so here](http://docs.anaconda.org/using.html)

The readers digest is that:

- you can have a package uploaded automagically if its build / test passes:
```bash
conda config --set anaconda_upload yes
```
- note that when a build is succesful that the last line of the ``conda build`` command is how to upload this to anaconda

- register a token such that you can upload and add pythonocc to your channels:

```bash
anaconda auth --create --name YOUR-TOKEN-NAME --scopes 'repos conda:download'
conda config --add channels https://conda.anaconda.org/t/TOKEN/USERNAME
```

rainman110, 2016-02-16:
> Is your oce-0.17.1 build against vtk ?

@tpaviot  Not yet. I'll integrate it in a next step. I hope, VTK is also available via conda.

jf---, 2016-02-16:
Succes!

[image]

rainman110, 2016-02-16:
Nice pace guys!

jf---, 2016-02-16:
cheers :tropical_drink: @rainman110 and @tpaviot , happy too about the progress being made...
really looking fwd using the OpenGL / GLSL shaders in 0.17... that should be exciting.

wrt VTK: 

- yep [there's a build](https://anaconda.org/anaconda/vtk). note that usually the conda recipe is part of the file, so if you want to do a custom build, don't start from scratch ;)

- all unit tests are passing... on 0.17 congrats @tpaviot nice work!

- I'll go through all the examples tomorrow ( tried 5 or 6 tonight all passing, so far so good )

jf---, 2016-02-17:
btw, I've got a decent built of [pythonocc 0.17 here](https://anaconda.org/jf/pythonocc-core/0.16.2/download/osx-64/pythonocc-core-0.16.2-py27_73.tar.bz2). Strangely enough the built ( utilizing the travis / conda script ) generates a wrong package name...

you probably know this, but its adviced to use this in a separate environment

```bash
conda create -n pyocc017 python=2 
```

jf---, 2016-02-17:
@tpaviot, @rainman110 , would it be possible to test the integrity of the wrapper can be checked automagically?

For instance, when a class is not wrapped succesfully, when you exec dir(theClassInQuestion) you get something like this as your output. ( how do you call this, just a SWIG pointer? ).

Would there be a way to find these cases without having to instantiate the classes nesecarily?
If that's the case, perhaps it could be automated. That way, it would be easier to check between different pythonocc / OCE version if all is wrapped correctly.

I'm assuming that if this would be possible at all, that Thomas would already have done so :sunglasses: hope you dont mind me asking...

```python
['__class__',
 '__del__',
 '__delattr__',
 '__dict__',
 '__doc__',
 '__format__',
 '__getattribute__',
 '__hash__',
 '__init__',
 '__module__',
 '__new__',
 '__reduce__',
 '__reduce_ex__',
 '__repr__',
 '__setattr__',
 '__sizeof__',
 '__str__',
 '__subclasshook__',
 '__weakref__',
 '_kill_pointed',
 'this',
 'thisown']
```

rainman110, 2016-02-17:
> btw, I've got a decent built of pythonocc 0.17 here. Strangely enough the built ( utilizing the travis / conda script ) generates a wrong package name...

@jf---  I think this is a bug in conda if you are working on a separate branch or fork. I had this bug quite often already. To determine the version number, conda tries to figure out, on which (latest) tag the current commit is based. Obviously, this does not work very reliable.

rainman110, 2016-02-18:
@jf---  What would be the benefit? 

If wrapping fails, we'll get a build error. 

If we forget some method to include due to a faulty generator, we have no way to determine this.

The best way to test integrity is to have a large collection of unit tests, IMHO.

jf---, 2016-02-18:
hi @rainman110 ,

I think we are on the same page -- what I'm thinking off is whether we can think of a script, that detects where the wrapper did not succeed wrapping the API. 2 posts above, I tried to describe what this looks like.

This is useful for integrating new features from different OCE / OCC versions.
This way, the developer would be informed of corner cases by running a script that detects these cases. 

Thing is that we can have a correct build, without the wrapper actually being functional.
[See here](https://github.com/tpaviot/pythonocc-core/issues/223#issuecomment-185570613), where a new idiom in the code base is not covered by the wrappers.

tpaviot, 2016-03-01:
@rainman110 After I merged the new ms branch, I rebased the branch review/tmp-port-to-oce-0.17.1 on top of master, regenerated the SWIG files. travis still complains about an issue with the GUI test under Linux (damned!), this time the line ```myV3dView->SetWindow(myWindow);```of the file Display3d.cpp seems to raise a segfault.

rainman110, 2016-03-01:
@tpaviot Does this have to do with the new memory management? I.e. did the code work before merging with the new memory management?

I just ask, because the code creating the graphic driver and the display connection in Display3d looks awkward ( at least, I implemented it in a different fashion in TiGL Viewer).

tpaviot, 2016-03-01:
@rainman110 Nothing to do with the new memory management, the bug was already there before I merge the ms branch. But it dit not solve it ;)

tpaviot, 2016-03-02:
@rainman110 Ideed the Display3d.cpp code was buggy, thanks for pointing me in the right direction. I pushed a fix. According to travis, it works on Linux, I still have though to check if it does not break the Windows behaviour.

tpaviot, 2016-03-02:
ok, that works on windows too

rainman110, 2016-03-02:
@tpaviot  Nice to hear!

What remains to be done for the 0.17 release? Everything in good shape?

tpaviot, 2016-03-02:
@rainman110 I'd like to test the vtk integration. And maybe we can release a 0.16.4 that includes the new memory management

jf---, 2016-03-02:
hi @rainman110 

wrt 0.17:
  
1. there's been a lot of work on OpenGL, working towards a non FFP implementation ( GLSL shaders work, you can have Qt decide when to swap the OGL buffer, GLSL warnings are optionally caught ). I'll put out a PR for this today.

2. not all examples are passing

3. some new C++ idioms are introduced that are not automagically wrapped. this is tangent to 1 , TBD on the PR

4. a GLSL example would be important ( the groundwork for that is laid out, again PR tonight )

rainman110, 2016-03-02:
Regarding the new memory managament. Is there any larger software that uses pythonocc to a great extend?

I would be good to have those developers doing some beta tests. The new memory management is a really huge change which should be thouroughly tested. A real world code can probably find some gotchas that I / we didn't think of yet.

jf---, 2016-03-02:
hi @rainman110 ,

So I'm using pythonocc to develop robotics software. Its a fairly extensive project ( ~40k SLOC ).
At the moment I'm refactoring the GUI to use QML, for which I need 0.17 ( since now you can have Qt control the OpenGL buffer, this is a technical requirement for QML ) 

The project has decent test coverage, so that might help.
That said, I'm not sure whether this refactor will be completed when we'd like to release 0.17.
I think there's still some work ahead of us...

rainman110, 2016-03-02:
hi @jf--- The qml development sounds really exciting. I am looking forward to see some implementation.

Did you also try out the 0.16 release with the changed memory management? If yes, did everything work as it should work?

rainman110, 2016-03-12:
@tpaviot I added VTK support for the oce-0.17 conda builds. I already uploaded binaries for OS X and Linux.

It would be nice to have some VTK example in pythonocc :smile:

tpaviot, 2016-03-14:
@rainman110 Great. Did you build oce-0.17.1 or the latest master including VTK changes ?

rainman110, 2016-03-14:
@tpaviot I built oce 0.17.1. I had to add a patch to be able to compile it.

tpaviot, 2016-03-14:
@rainman110 Ok i think a 0.17.2 release is necessary, to avoid patching 0.17.1 whenever VTK support is needed. I will do that in the next coming days.

tpaviot, 2016-03-16:
@rainman110 FYI I started working on the vtk support for pythonocc. travis fails, as usual ;)

kremnef, 2016-04-01:
Hi @tpaviot , I am interested in pythonocc. How's it going with the oce 0.17?

jf---, 2016-04-01:
@kremnef , you can use the [review/tmp-port-to-oce-0.17.1](https://github.com/tpaviot/pythonocc-core/tree/review/tmp-port-to-oce-0.17.1) branch. Its still WIP...

kremnef, 2016-04-02:
@jf---  thanks a lot) 
_configuration_ is ok, but _make_ has errors:

```
@Kremnevs-MacBook-Pro:cmake-build kremnevandrey$ make
[  0%] Swig source
Scanning dependencies of target _XCAFPrs
[  1%] Building CXX object CMakeFiles/_XCAFPrs.dir/Darwin/x86_64-MinSizeRel-64/XCAFPrsPYTHON_wrap.cxx.o
[  1%] Linking CXX shared module Darwin/x86_64-MinSizeRel-64/_XCAFPrs.so
Undefined symbols for architecture x86_64:
  "Graphic3d_Vector::Graphic3d_Vector()", referenced from:
      __GLOBAL__I_a in XCAFPrsPYTHON_wrap.cxx.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [Darwin/x86_64-MinSizeRel-64/_XCAFPrs.so] Error 1
make[1]: *** [CMakeFiles/_XCAFPrs.dir/all] Error 2
make: *** [all] Error 2 
```

jf---, 2016-04-02:
@kremnef, [this PR](https://github.com/tpaviot/pythonocc-core/pull/233) fixes this issue...

crobarcro, 2016-04-18:
Hi, I've just had a go building the branch `review/tmp-port-to-oce-0.17.1`, but I'm getting the following error:

```
[ 95%] Built target _V3d
[ 95%] Swig source
Unable to open file /home/rcrozier/build/pythonocc-core-git/src/Visualization/VisualizationPYTHON_wrap.cxx: No such file or directory
make[2]: *** [src/Visualization/VisualizationPYTHON_wrap.cxx] Error 1
make[1]: *** [CMakeFiles/_Visualization.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Linking CXX shared module Unix/x86_64-MinSizeRel-64/_Visual3d.so
[ 95%] Built target _Visual3d
make: *** [all] Error 2

```

I'm using the 17.1 version of oce taken from the FreeCAD maintainers ppa. I'm also using SWIG 3.0.7. I'm on mint linux. Is it my fault (likely) or a build issue?

crobarcro, 2016-04-18:
I switched to swig 2.0.11 and the build completes fine!

gekkoman, 2016-05-16:
Switched to 0.17 tree since arch just went  to oce 0.17 breaking python-occ. 

I am new to python-occ and opencascade so it may not be a 0.17 issue. 

No issues installing and running. Everything was going swimmingly until trying to use selected object callback return  out of  qt5 visualisation. (AIS_InteractiveContext with AIS_Shapes). 

It mostly works but if you select a shape which shares a TShape handle with another shape the GUI returns all shapes sharing the same TShape (eg behavior like  IsPartner() not IsSame()) leaving no easy way to discriminate correct selected object. The rendering however displays the correct as expected highlight.

Even more perplexing:  

```
obj = someAIS_Shape()
context.Display(obj.GetHandle())
```

then inside callback 

```
def click_xy(self,shp, *kwargs):
    for shape in shp:
       obj.isSame(shape)
```

fails if the correct  object obj is selected. 

It appears here that  the Location() data is failing equality check.

Only spotted this because to the multiple TShape return....most of the time a single correct object is returned.

I have iterated through the shape location data (eg location, datum, power etc) and it appears identical for obj and shape  but fails equality check. Upstream issue?

gekkoman, 2016-05-16:
The select issue from above is upstream....you are simply returning the AIS.InteractiveContext selection. Please ignore and I will post in a more appropriate spot.

tpaviot, 2016-05-25:
I just updated the review/tmp-port-to-oce-0.17.1 branch and renamed it to review/port-to-oce-0.17.1

tpaviot, 2016-06-04:
The review/port-to-oce-0.17.1 has been merged into master. I close this issue.

closed 2016-06-04 by tpaviot

Vishnu-C, 2018-07-26:
Is there a example for VTK rendering support?