tpaviot/pythonocc-core #214: Discussion: Memory management

rainman110, 2016-02-01:
**This "issue" is aimed to be a discussion thread. I tried to write down my findings about potential problems of the current memory management. It should help me and others to understand, why the memory management is designed how it currently is.**

Memory management of the OCCT classes is somewhat difficult due to the ```Handle``` classes.

Handles take care of the life time of an object. If an object does not have any references to itself anymore (in terms of handles), it will be automatically deleted. Handles are basically smart pointers.

There are still memory issues left in pythonocc, most of them leaks.

Therefore I tried to understand, how the memory management is currently implemented in pythonocc and were are the problems originating from.

This brought some basic questions/issues:

  1. Why is the garbage collection required? There are other ways to deal with smart pointers and ownership of pointers than garbage collection.
  2. The garbage collection basically prevents the deletion of objects and shifts it to a later time. This brings up several problems, e.g. memory leaks if you forget to use the garbage collector.
  3. The way how the methods ```GetHandle``` are implemented looks strange. One particular issue with this approach is, that you could create a Handle of an object that lives on the stack. This only works in pythonocc currently, because you implicitly assume that swig always allocates objects on the heap. If a handle points on a stack object, it will try to delete its parent object, which will lead to a segfault. Why aren't handles create as in OpenCASCADE i.e. : 
 
        myHandle = Handle_Geom_BSplineSurface(Geom_BSplineSurface(...))

Because of these "issues", I played around with some toy project to better understand the memory management. It is completely possible to wrap the Handle* classes without any garbage collection and without any memory leaks.

When creating a handle to an object, the handle takes ownership of the lifetime of the object. At this point, also python/swig already has ownership of the object. Normally, this will lead to segfaults, as both python/swig and the handle try to delete the object. By using 

     MyObject.thisown = False

it is possible to remove the ownership from python/swig. In swig, the change of an ownership can be even defined in the interface file using code like:

```c++
%apply SWIGTYPE *DISOWN { A *adisown };
    
class Handle_A
{
public:
  Handle_A(A* adisown) {
   a = adisown;
   if (a) a->IncRefs();
  }

private:
A* a;
}
```
Unfortunetly, this feature is hardly documented in swig.

tpaviot, 2016-02-02:
@rainman110 Thanks for opening this thread. I spent hours working about memory management issues in pythonocc.

OCCT Handles are basically smart pointers I agree. Unfortunately I never succeeded to wrap this smart pointers mechanism using SWIG, despite the fact SWIG can work with boost smart pointers. As a consequence, there's a continuous fight between python and c++ to decide who shloud delete the object. Indeed, both python and oce try to delete the object in different ways.

1. About the garbage collection: before the garbage collection collectors (for transients, handles, ohers), pythonocc used to segfault. For instance, if you create some topods_* instances in a function, build a primitive on top of these topods_*, and exit the function, then the topods_* instances get deleted. Many primitives use references to objects, other don't.

That's why I removed ownership for python objects, try to collect them whenveer the __del__ method is called, and then really delete them "when needed", which means "later", or often "never I don't care if objects are not deleted, memory will be cleaned when the script stops" ;-)

So the current situation is: pythonocc works in the sense that you can call every method/function provided with oce/occt. pythonocc fails at properly manage memory that's absolutely true. So far, each time I had to deal with big objects or heavy data weights, I always used batch processing. Divide ad conquer. It's much quicker than a single thread, easier to develop than multithreading, safe in terms of memory management. But if you develop some kind of GUI application using pythonocc, working long lifecycle data (I mean minutes or even hours), then this problem cannot simply be ignored.

Why is the garbage collection required ? To be honest, I did not manage to get any of the other solutions working. Dealing with c++ and memory management is just a nightmare IMO, mostly because I'm not a c++ specialist. That's exactly why I started pythonocc: I didn't want to use open cascade from c++! Working with both c++/swig/python requires an excellent knowledge in both these three languages: SWIG is really hard to understand (I spent hours reading the generated c++ code trying to understand what's hapening), and I'm a poor c++ programmer. Here's the origin of the "garbage collector" ;-)

2. I fully agree. Even if you use the garbage collector, you'll get surprises/segfaults because the smart_purge function is not as smart as its name suggests. I should have named it "dumb_purge".

3. If you have an idea, or if you successfully experiment better solutions, perfect. Just tell me what to do, I'll update the wrapper generator to fit your suggestion. For instance, copy/paste here a class as it should be defined in the SWIG .i file.

rainman110, 2016-02-02:
@tpaviot Thank you for the detailed explanation. This helped me a lot to understand the backgrounds of the memory management.

I'll try to reproduce the issues with the TopoDS_Shapes and primitives. When I completely understand the issues, I will try to find a concept without garbage collection.

>  For instance, if you create some topods_* instances in a function, build a primitive on top of these topods_, and exit the function, then the topods_ instances get deleted. Many primitives use references to objects, other don't.

@tpaviot Can you please specifiy in more detail, what kind of topods_* instance you are creating and what primitives are you making? I could imagine, creating some TopoDS_Edge(s) and the make a wire using BrepBuilderAP_MakeWire. Would this (pseudo) code suffice to demonstrate the original problems? :

```python
def get_wire():
    e1 = BrepBuilderAPI_MakeEdge(p1, p2).Edge()
    e2 = BrepBuilderAPI_MakeEdge(p2, p3).Edge()
    wire = BrepBuilderAPI_MakeWire(e1, e2).Wire()
    return wire

myWire = get_wire()
```
My main programming language is C++, thus our knowledge should complement each other. I am not that strong on the python side though. The generated swig code and also swig itself is a bit nasty, but it's a working horse and does it's job quite well. I hope, that I can contribute a working solution in the end.

tpaviot, 2016-02-02:
It's been a long time ago I did not work on the memory management issue. I can't remember exactly which kind of objects/methods to call in order to raise a segfault. I guess your ```get_wire``` may be a good test, but the only to get sure about that is to modify the TopoDS.i swig file, and give back to python the object owning (replace self.thisown=False with self.thisown=True everywhere). You'll get a pythonocc working without garbage collector.

A few other things related to memory management:
* oce/occt allocates/deallocates memory according to the MMGT_OPT env var value. It can be 0, 1 or 2. Have a look at the Standard.cxx file. 0 stands for "default", 1 for "optimized" and 2 for "TBB". I noticed different behaviours of my smart_purge method: according to MMGT_OPT, segfault or not, memory released or not.

* In an ideal world, python users should not have to use any Handle_*. It's a c++ specific way to handle objects, but it's not pythonic at all. I wish we can make these Handles hidden to python users.

rainman110, 2016-02-03:
I might have an idea, where the original segfault problems were coming from.

Please have a look into this pseudo code:

```python

def do_something(localA):
    # get handle to a to compute something
    # from now on, a_handle takes care of a's lifetime
    a_handle = localA.GetHandle()

    # a_handle has now ownership to a
    # we need to release the ownership from python
    localA.thisown = False

a = A()
do_something(a)

# the next line will cause a segfault as a will destructed 
# when a_handle gets out of scope
a.compute_something()
```

This code demonstrates, what kind of problems occur when creating a handle to an object. As the handle will take care of a's lifetime, a is invalid when it's handle becomes invalid.

This is clearly something we don't want. In pythonocc, we need handles only as input arguments to make some functions happy.

I have a rough idea, how we could solve this issue. We could accociate/integrate a handle object into the shadow class of each transient object. When we call ```GetHandle```, we simple return this handle object. When our transient object becomes invalid (i.e. goes out of scope), it will also delete its handle which automatically calls the destructor of a.

What do you guys think about this approach? I will try to make a small demo to test this idea.

rainman110, 2016-02-03:
@tpaviot I created a toy project on github to test my ideas wrt to the memory issues.

I implemented my ideas and I have no issues so far. Meaning, no crashes and no leaks.

The result can be seen here: https://github.com/rainman110/swighandles

The main idea was, to add a handle class to each shadow class. This handle takes care of the life time tracking and is also used for the getHandle method. Please let me know what you think!

tpaviot, 2016-02-03:
That's very interesting. Your experiment can easily be ported to pythonocc-generator, we should see how it runs. And if it succeeds where the current implementation fails (segfaults/leaks). Let's create another brancn on pythonocc-core repos to test that

jf---, 2016-02-03:
exciting to see this move fwd... :tropical_drink: 
looking fwd to see this applied to the pythonocc-generator indeed!

rainman110, 2016-02-04:
I implemented my ideas to the wrapper generator. Unfortunately, I have some problems with c++ functions returning a handle like:

```python
myCurveHandle = GeomAPI_PointsToBSpline(...).Curve()
# now, the next line already crashes
myCurveHandle.GetObject().NbKnots()
```

I analyzed whats happening. It seems, that the curve object (not the handle) is destroyed already in the first line. GeomAPI_PointsToBSpline returns the handle and copies it to myCurveHandle. This already triggers the destructor of Geom_BSplineCurve, which should not happen actually.

I'll have to dig further into it, to find out what actually happens here.

here's my work: https://github.com/rainman110/pythonocc-generator/commit/1613fa88cdd2230c70c7a9ce7b6113cd148f36bf

rainman110, 2016-02-05:
After hours of valgrind sessions, I narrowed the problem down. It has to do with object lifetimes in python.

The minimal example for this problem:

```c++
class D
{
public:
    D() {v = 4;}
    int GetValue() {return v; }

private:
    int v;
};

class C
{
public:
    C() {d = new D();}
    ~C() {delete d;}

    D* GetD() { return d; }
private:
    D* d;
};
```

And this is the python code:

```
# this line is working incorrectly
print C().GetD().GetValue()
>>> 7036080

# this is working on the other hand
c = C()
c.GetD().GetValue()
>>> 4
```

The reason for the issue is, that in first line ```print C().GetD().GetValue()``` the C() object is destroyed before call GetValue. The destructor of C also deletes its D object, which makes the call to GetValue invalid.

We need a way, to tell swig, that there is a lifetime dependency between D and C when calling C.GetD. Swig must know, that as long D exists, C must not be garbage collected. @tpaviot  Do you know about a swig mechanism that allows this?

rainman110, 2016-02-05:
I found a way to circumvent this upper error. It is a bit hackish but it works.

We do not return the object by pointer or reference, but by value:

```
%extend C {
  D GetD() {
      return $self->GetD();
  }
}
%ignore C::GetD();
```

In pythonocc, D would be typically Handle_Geom_BSplineCurve. GeomAPI_PointsToBSpline returns the handle by reference. With this trick, we can force it to be returned by value. 

In order to implement this, we have to search for classes, that return a Handle by reference or by pointer. Then, this method must be disabled and rewritten using the %extend mechanism.

This is however not very elegant, but I can not think of anything other.

By the way, the std_sharedptr typemap of swig suffers from the same problems. This is a limitation of swig and python.

The other solution would be, to add all objects to a garbage collector like we currently do. We just don't need to care of, what kind of object we are collecting and don't need to do dangerous things like kill-pointed.

What do you think?

tpaviot, 2016-02-05:
@rainman110 Your example 

```python
myCurveHandle = GeomAPI_PointsToBSpline(...).Curve()
# now, the next line already crashes
myCurveHandle.GetObject().NbKnots()
```

perfectly illustrates what lead to the garbage collection solution I chose. By removing any call to a destructor, it magically let pythonocc performs any operation, without having to take care about segfaults/objects destruction.

I think there is an issue with the way Transient objects are created/allocated. According to the c++ code I read, all those objects should be created using a Handle, for instance:
```c++
Handle(TColGeom_HSequenceOfCurve) sequenceOfCurve = new TColGeom_HSequenceOfCurve();
```
It's not the way it's done in pythonocc. I will look at the generated c++ file but I'm quite sure it's not like that.

I'm very surprised to read that the std_shareptr typemap suffers from the same problem.

Your suggestion to return the Handles by value rather than by reference or pointer is interesting. It has to be tested, for instance for a few classes first. I remember playing a lot with those kind of things a long time ago, not the best period of my life ;-)

The garbage collector as it is now has to be improved. Objects are properly collected, but I didn't find any safe and clean way to destroy the objects. In order to be efficient, the purge should be automated, since it's not very pythonic to call a garbage purge manually. I wonder if it is possible. Anyway, both solutions are tricky. But I think that any working solution, even if inelegant, is better than a garbage collector.

I also remember trying to use the python gc module without success.

rainman110, 2016-02-05:
> Your suggestion to return the Handles by value rather than by reference or pointer is interesting. It has to be tested, for instance for a few classes first. I remember playing a lot with those kind of things a long time ago, not the best period of my life ;-)

I hand-fixed some of the swig .i files. For the fixed methods, this trick seems to work and the geometry unit tests are functioning.

I could try to integrate this into the generator, to have this path applied at large scale.

> The garbage collector as it is now has to be improved. Objects are properly collected, but I didn't find any safe and clean way to destroy the objects. 

With my previous idea with the handle integrated into the transient object, object destruction should be trivial. Just call the appropriate delete_MyClass destructor of the swig generated _module.

> I think there is an issue with the way Transient objects are created/allocated. According to the c++ code I read, all those objects should be created using a Handle, for instance...

I don't think that this currently makes any problem. We basically have this done with my suggestion presented in the swighandles demo.

rainman110, 2016-02-05:
Next status: I implemented my idea  "return by value Handle_" into the generator. Instead of using extend, it could be done even simpler:

a line

```
const Handle_MyClass& GetValue();
```

had to be changed to

```
Handle_MyClass GetValue();
```

The rest is done automagically by swig.

This way almost all tests are running correctly. The only exception is the Tesselator class. Here I have to look into it, why all Tesselator examples are crashing now :(

tpaviot, 2016-02-06:
Only one line change should be easy to implement in the generator.

The Tesselator class is a hack over the officiel occt, it might not be as clean as the original code.

rainman110, 2016-02-07:
Actually, I already integrated the change into the generator locally. I could push it into my repo and you can check this when you have time.

Regarding the tesselator. I didn't have the time to analyze the origin of the problem, but I don't think, that the issue lies in the tesselator itself. I'll do a valgrind session when I have time.

rainman110, 2016-02-07:
Here's the commit with the "Return by value" implementation. Nothing very special...
https://github.com/rainman110/pythonocc-generator/commit/1f08053c03ffb8268fdc12286d0af19e6619aea6

tpaviot, 2016-02-08:
Ok, I will create a branch with your patch on pythonocc-generator and try to remove the garbage collection to see what's going on. I won't update the .conf file, specific to each user.

rainman110, 2016-02-08:
Yes, sorry for the conf file. I committed is accidentally. I forced push a new commit without the changes to the conf file.

tpaviot, 2016-02-08:
@rainman110 I pushed your patch to tp/return-handle-by-value at pythonocc-generator. All the tests/example pass, including the one that involve the Tesselator class.

tpaviot, 2016-02-08:
@rainman110 Together with your first commit, it seems to work. I mean tests/examples pass without any garbage collector. Kudos !

BTW, no problem with the Tesselator class. We need travis-ci to be fixed to check everything is ok on linux/osx.

I'll run a few more tests before creating a pull request on both pythonocc-generator/pythonocc-core.

tpaviot, 2016-02-08:
I ran the following example

```python
from OCC.BRepPrimAPI import BRepPrimAPI_MakeTorus

while True:
    shp = BRepPrimAPI_MakeTorus(40, 30.).Shape()
```
Monitoring memory connsumption at the same time, using the Windows task manager, it appears that the memory grows up and is never released.

rainman110, 2016-02-08:
I just checked at a small example, that the number of destructor and constructor calls of Geom_BSplineSurface/Curve are identical. There it works. I'll have a look onto your example.

We'll figure out whats happening there.

rainman110, 2016-02-09:
@tpaviot I analyzed your example. The increase in memory usage is a python + swig problem due to some internal variables not freed.

I made an analysis of the original pythonocc and again with the new modified one. Here's the valgrind diff between both: http://pastebin.com/PMiXGHYV

As you see, all OCC specific allocs are freed now with the new memory model.

rainman110, 2016-02-09:
@tpaviot I simplified the new memory manament even further. Now, all tests are working except from the memory management tests,  obviously.

Valgrind shows a general improvement in memory usage. No garbage collector is required anymore.

The changes to the generator a now minimal:
https://github.com/tpaviot/pythonocc-generator/compare/master...rainman110:ms/mem_mgr_revamp?expand=1

The interface file are also extremely simplified:
https://github.com/tpaviot/pythonocc-core/compare/master...rainman110:ms/new_mem_management?expand=1

Now, I have a good feeling that these changes are solid and can integrated into master as soon if you want.

I can provide some testing conda builds, if you want. Somehow, the problems with the tesselator class vanished. I am not sure what was happening there. No problems at all anymore.

tpaviot, 2016-02-09:
@rainman110 I agree that this can be merged asap. Please open a pull request on pythonocc-generator.

rainman110, 2016-02-09:
@tpaviot What are we doing with the memory management tests? They fail, which will make the ci build fail.

Do you want to remove the tests completely or should I modify it to account for the new code?

tpaviot, 2016-02-09:
I will create a new branch on pythonocc-core, review/new-ms, I will delete the GarbageCollector.py file and the related tests as well

rainman110, 2016-02-09:
@tpaviot I would not delete the garbage collector, the maintan client code compatible. We just need to change the code to do nothing ;)

tpaviot, 2016-02-09:
@rainman110 you're right, although I doubt anyone ever imported the garbage collector. So let's modify the code so that a "Deprecation Warning" is raised whenever the Garbagecollector is imported

rainman110, 2016-02-10:
@tpaviot  My current implementation has one flaw left. It crashes with the following code:

```python
def evil_function(t):
	handle = Handle_Standard_Transient(t)
        # when using the next line instead, everything is okay
        # handle = t.GetHandle()

t = Standard_Transient()
evil_function(t)
```

This is the C++ way of creating handles and we can't stop anyone to use this syntax. Therefore, I suggest to extend the constructor of the Handle classes, to register the handle into the base object. I am currently working on the implementation.

tpaviot, 2016-02-10:
@rainman110 Fine, push an update to your PR onpythonocc-generator, I'll update the related branch on pythonocc-core. The evil_function test must be added to the unittest suite, I'll do that.

tpaviot, 2016-02-10:
@rainman110  on my machine, the evil_function does not produce any crash (mingw compiler)

rainman110, 2016-02-10:
@tpaviot This is strange! On mine (Visual C++ 2008) it crashes. The crash totally makes sense, as the handle destroys t and the python also tries to destroy t.  This leads to a double free corruption.

jf---, 2016-03-29:
Gents, since the memory refactor is merged, can I aks @tpaviot  to close this?

rainman110, 2016-03-29:
Yes, of course. I could also close this, but I'll let @tpaviot decide.

closed 2016-03-29 by tpaviot