tpaviot/pythonocc-core #222: Can you point me please to the high level API?

dcowden, 2016-02-13:
Hi, Guys:
I've been away from pythonOCC for a long time, and things have changed quite a bit.

Does pythonOCC offer a standard 'high level' API these days? If so, where could i find it?

Currently a library I have written, cadquery, offers really fluent scripting to make solids. it is indirectly based on OCE-- but currently uses FreeCAD. Here are some examples:

http://dcowden.github.io/cadquery/examples.html

We are soon going to be moving cadquery to use OCE directly. There is a lot of discussion about whether to use python or javascript for CQ 2.0-- but if we use python, we'll certainly use pythonOCC.

I don't want to re-invent the wheel, so if pythonOCC has any standard, high level constructs, or plans to implement them, I definitely want to use them, before I dig in and do things from first principles.

Thanks for the help!

jf---, 2016-02-17:
hi @dcowden , happy to have you back...
sorry to take a while to get back to you.

so Guillaume Florent has done work is refactoring the [high level API](https://github.com/tpaviot/pythonocc-utils/tree/master/OCCUtils) I've been working on.
i think that's a good start, and a lot cleaner than the messy state OCCUtils is in.
it would be great to have these efforts converge.

Charles Sharman [ccad](https://github.com/charles-sharman/ccad) has been working on a more procedural API, there's lots of great things to be had there.

Finally, in welcoming you back, I'm happy to report there's been a lot of great work in moving pythonocc to conda for building / distributing it. This is a huge deal if you aks me, since the barrier to entry has lowered considerable. 

We're working now on an "official" pythonocc anaconda page, for now I'd like to direct you to Martin Siggel's page: https://anaconda.org/DLR-SC

Good to have you back :+1:

dcowden, 2016-02-17:
hey, thank you!
I'll review the high level api. i'm so hopelessly behind, i dont know what
anaconda even is.  But i'll learn shortly.

@Jelle, can you provide high level thoughts on the high level api that
exists now vs cadquery ( http://dcowden.github.io/cadquery/examples.html )

My mission is to figure out how best to avoid re-inventing the wheel, while
getting the API coolness I'd like

On another note ( shamelessly hijacking this thread):  I have an unrelated
question.

With cadquery, one of the big concepts was selectors.  I've been trying to
figure out how to implement a specific type of selector: 'created by'.

The use case is as follows. Suppose you start with a cube, and then you cut
a hole through the top and bottom faces by subtracting a cylinder.   In
this case, a cylindrical face is created inside of the cube as a result of
the subtraction.

What i want to implement is a way to identify the edges, faces, and bodies
that were created as a result of a given operation. In this case, the
subtraction created two edges and one face.

How could i do this using pythonOCC?  TThe only way I could figure out how
to do it is really nasty. I could compute a hash of each edge, face, and
body in the original ( here, the cube). then, after the operation, i can
iterate over all of the features, comparing hashes again. If they match, it
is the original feature, and thus it was not created by the current
operation. But if it is new, it was created by the new feature.

Is there a better way?

closed 2017-05-04 by tpaviot