tpaviot/pythonocc-core #223: enabling raytracing

jf---, 2016-02-17:
Here is a minimal example, that tries to enable raytracing.
However, currently, rendering parameters cannot be altered.
Graphic3d_RenderingParams is probably not wrapped correctly.

```python
from __future__ import print_function

from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox, BRepPrimAPI_MakeSphere
from OCC.Display.SimpleGui import init_display
from OCC.Graphic3d import *

display, start_display, add_menu, add_function_to_menu = init_display()

ais_boxshp = None

def build_shape():
    boxshp = BRepPrimAPI_MakeSphere(50.0).Shape()
    ais_boxshp = display.DisplayShape(boxshp, update=True)
    return ais_boxshp

build_shape()

render_params = Graphic3d_RenderingParams()

display.View.ChangeRenderingParams(render_params)

>>> TypeError: V3d_View_ChangeRenderingParams() takes exactly one argument (2 given)

start_display()

```

jf---, 2016-02-17:
a bit of context from ``oce/samples/qt/Common/src/View.cxx``

```c++
  if (myIsRaytracing)
    myView->ChangeRenderingParams().Method = Graphic3d_RM_RAYTRACING;
}
```

tpaviot, 2016-02-18:
ChangeRenderingParams is a c++ method that changes params by reference. The Graphic3d_RenderingParams is a bit special and needs a specific wrapper.

tpaviot, 2016-02-18:
This is a hack to change the rendering method. In the file Graphic3d.i, add the following method to the Graphic3d_RenderingMethod (around line 5500):
````
%extend Graphic3d_RenderingParams {
	void SetRayTracing() {
		$self->Method = Graphic3d_RM_RAYTRACING;
	}
};
```
Recompile, and then modify your script as follows:
```
display.View.RenderingParams().SetRayTracing()
```

tpaviot, 2016-02-18:
You can alse hack the Display3d class, to experiment:
```
myV3dView->SetWindow(myWindow);
++ myV3dView->ChangeRenderingParams().Method = Graphic3d_RM_RAYTRACING;
```

jf---, 2016-02-18:
Yes, its really not very idiomatic code.
I like your idea of doing this in the Display3d class rather than is SWIG code for this reason.
Will look into this, thanks for the pointer Thomas...

jf---, 2016-04-08:
will adress this in PR #238

closed 2016-04-08 by jf---