tpaviot/pythonocc-core #224: Display get_backend Import error in Spyder

p-chambers, 2016-02-23:
Hi,

I have been using PythonOCC from the Spyder IDE for a while, and have always closed the Qt window and rerun the script and restarted the display when I want to remake the geometry - However with some recent changes, the get_backend function in /Display/backend.py throws an error on the second run as the IPython kernel has already called get_backend, forcing me to restart the kernel. I initialize the display with:

from OCC.Display.SimpleGui import init_display
display, start_display, add_menu, add_function_to_menu = init_display()

start_display()

Should I be changing my workflow and not calling the usual display initalizer on each run, or is this an IPython/IDE-related bug? 

Thanks

jf---, 2016-02-24:
hi @p-chambers ,

Lets see... now, I dont have spyder running, is it possible for you to reproduce the issue with say, ipython? or write a small script such that we can reproduce the problem?

If I follow along, something amongst these lines happens:

- first, you call start_display, in a fresh interpreter. all is peachy.
- now you restart the same script. 
- so there a GUI loop already running?
- normally, in a Qt app, you cannot interrupt this GUI loop you see... 

I'm pretty sure something can be figured out.
You know what, one of my favorite tricks is __not__ calling ``start_display``
I do this when developing a large app, with lots and lots of dependencies ( -> long loading time )
With Ipython's reload functions, the app is relaunched instantaniously, with the edited modules reimported... saving a lot of time...

Hope that helps for know... if you can help to reproduce the issue that would mean a lot for us...

Oh, btw, are you aware of efforts [ipython integration in pythonocc-contrib?](https://github.com/tpaviot/pythonocc-contrib/tree/master/WebServer) ( I agree "webserver" is perhaps somewhat misleading, though technically correct ;) )

p-chambers, 2016-02-26:
You are mostly correct there, however I used to close the Qt window thus freeing the ipython console to rerun the script and generate a fresh geometry in a fresh window. You can reproduce the issue with any of the core Qt examples and running twice in a standard ipython interpreter with the %run magic e.g. 

> %run core_geometry_bspline.py        #(x2)

In fact, I would previously use the %gui qt magic first, allowing me to just rerun the script and the Qt window would restart on its own and the ipython interpreter would remain free ... This made things nice and easy in Spyder as running a file is bound to a keystroke, so I could always reproduce the same output with a button (perhaps comprising efficiency).

 If it helps, I am using Anaconda Python and the prebuilt Conda package from the DLR repository.

Yes I have seen and am using the ipython notebook webserver alongside my spyder endeavours - here's a screenshot :).
[image]

tpaviot, 2016-04-06:
@jf--- This regression is related to recent Qt5 changes. Could you please fix #224 and #225

jf---, 2016-04-06:
hi @tpaviot , #225 is a regression.
the unexpected behaviour here is due to the eventloop of the GUI not being properly managed.
this is beyond the scope of pythonocc-core's implementation.
I'm in favor of closing this issue, its too vaguely defined, too ipython specific ( qt support != ipython support )

jf---, 2016-04-06:
( sorry to go offtopic... )

---

@tpaviot @p-chambers i think your [blog](https://aircraftgeometrycodes.wordpress.com/) is really very interesting... it would be cool to have the project on the pythonocc wiki / mention it in a pythonocc blog post or something... are you planning on a specific post on [OCC_AirCONICS](https://github.com/p-chambers/OCC_AirCONICS)?

p-chambers, 2016-04-08:
@jf--- @tpaviot in short, yes I am planning a blog post. I've sent you an email to avoid an off-topic discussion.

For the issue at hand: I suppose the answer then is that Ipython/IDE development is not supported, so I should switch to developing Qt viewer examples with a vanilla "python example.py" or just restart the Ipython kernel after each run?

Thanks

jf---, 2016-04-08:
hi @p-chambers , so we share a fascination for design applying evolutionary algorithms ;)
that's __exactly__ one of the reasons why I've been so interested in pythonocc too :sunglasses: 

---

so, ipython / qt integration... 
I've kind of explained how to go about this here above; the key is not calling `start_display`.
this start the event loop, which is the culprint -- you cant simple stop / start a Qt app's event loop.
ipython already starts an event loop ( ipython --gui=qt5 ) so calling start_display is not required anyway...

@p-chambers , who not try that and see where we can go from there?
actually, this way of working is really useful -- with ipython's autoreload you can keep on developing in your IDE and introspect the app through ipython...

tpaviot, 2016-04-24:
@p-chambers can you please have a look at #256, I think it may solve the problem with Spyder

closed 2016-04-25 by tpaviot

p-chambers, 2016-04-25:
Thanks @tpaviot, yes #256 seems to make things work as they did before. I'm not sure of the intricacies of how the IPython QT loop is managed, but it certainly seems to restart/reload the QT app again in spyder when I run user examples. Hopefully this will fall through into the DLR Conda build soon as this doesn't seem to have changed recently?

PR #256 (2016-04-25): qt backend fix
* the ```get_backend```function name is replaced with ```load_backend```, which is more explicit

* if the ```load_backend``` function is called more than once, the second times it returns an info message and the string of the load backend.

The behaviour is now:

```
>>> load_backend("qt-pyqt4")
INFO:OCC.Display.backend:backend loaded: qt-pyqt4
'qt-pyqt4'
>>> load_backend("qt-pyqt5")
INFO:OCC.Display.backend:The qt-pyqt4 backend is already loaded...``load_backend
y be called once per session
'qt-pyqt4'
>>>
```

Should fix #224