tpaviot/pythonocc-core #225: the PyQt4.QtCore and PyQt5.QtCore modules both wrap the QObject class

cristobaltapia, 2016-02-27:
I installed python-occ in Archlinux from here: https://aur.archlinux.org/packages/python-occ/ 
It built correctly but when I try to run the basix example in the web site ( http://www.pythonocc.org/quick-examples/hello-dumb-box/ ) I get the following error:

```
from OCC.Display.SimpleGui import init_display

from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox

display, start_display, add_menu, add_function_to_menu = init_display()
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-3-7053f85cee0b> in <module>()
----> 1 display, start_display, add_menu, add_function_to_menu = init_display()

/usr/lib/python3.5/site-packages/OCC/Display/SimpleGui.py in init_display(backend_str, size)
    139         if not app:  # create QApplication if it doesnt exist
    140             app = QtWidgets.QApplication(sys.argv)
--> 141         win = MainWindow()
    142         win.show()
    143         win.canva.InitDriver()

/usr/lib/python3.5/site-packages/OCC/Display/SimpleGui.py in __init__(self, *args)
     93             def __init__(self, *args):
     94                 QtWidgets.QMainWindow.__init__(self, *args)
---> 95                 self.canva = qtViewer3d(self)
     96                 self.setWindowTitle("pythonOCC-%s 3d viewer ('%s' backend)" % (VERSION, used_backend))
     97                 self.resize(size[0], size[1])

/usr/lib/python3.5/site-packages/OCC/Display/qtDisplay.py in __init__(self, *kargs)
    108 class qtViewer3d(qtBaseViewer):
    109     def __init__(self, *kargs):
--> 110         qtBaseViewer.__init__(self, *kargs)
    111         self._drawbox = False
    112         self._zoom_area = False

/usr/lib/python3.5/site-packages/OCC/Display/qtDisplay.py in __init__(self, parent)
     68 
     69         self._have_pyside = have_pyside()
---> 70         self._have_pyqt4 = have_pyqt4()
     71         self._have_pyqt5 = have_pyqt5()
     72         self._have_backend = have_backend()

/usr/lib/python3.5/site-packages/OCC/Display/backend.py in have_pyqt4()
     35     global HAVE_PYQT4, QtCore, QtGui, QtWidgets, QtOpenGL
     36     try:
---> 37         from PyQt4 import QtCore, QtGui, QtOpenGL
     38         QtWidgets = QtGui
     39         HAVE_PYQT4 = True

RuntimeError: the PyQt4.QtCore and PyQt5.QtCore modules both wrap the QObject class
```

jf---, 2016-02-27:
Looks like you have both PyQt4 & PyQt5 installed.
Either you can create a virtual environment ( conda! ) or explicitly set the backend to use ( eg init_display("qt-qt5")

tpaviot, 2016-02-27:
The backend python should take into account the case where both PyQt4 and 5 are installed, and take one or the other.

cristobaltapia, 2016-02-27:
Thanks for the tip. Unfortunately it doesn't seem to work:

```
display, start_display, add_menu, add_function_to_menu = init_display("qt-pyqt5")
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-3-daaf5d1bc5f5> in <module>()
----> 1 display, start_display, add_menu, add_function_to_menu = init_display("qt-pyqt5")

/usr/lib/python3.5/site-packages/OCC/Display/SimpleGui.py in init_display(backend_str, size)
    139         if not app:  # create QApplication if it doesnt exist
    140             app = QtWidgets.QApplication(sys.argv)
--> 141         win = MainWindow()
    142         win.show()
    143         win.canva.InitDriver()

/usr/lib/python3.5/site-packages/OCC/Display/SimpleGui.py in __init__(self, *args)
     93             def __init__(self, *args):
     94                 QtWidgets.QMainWindow.__init__(self, *args)
---> 95                 self.canva = qtViewer3d(self)
     96                 self.setWindowTitle("pythonOCC-%s 3d viewer ('%s' backend)" % (VERSION, used_backend))
     97                 self.resize(size[0], size[1])

/usr/lib/python3.5/site-packages/OCC/Display/qtDisplay.py in __init__(self, *kargs)
    108 class qtViewer3d(qtBaseViewer):
    109     def __init__(self, *kargs):
--> 110         qtBaseViewer.__init__(self, *kargs)
    111         self._drawbox = False
    112         self._zoom_area = False

/usr/lib/python3.5/site-packages/OCC/Display/qtDisplay.py in __init__(self, parent)
     68 
     69         self._have_pyside = have_pyside()
---> 70         self._have_pyqt4 = have_pyqt4()
     71         self._have_pyqt5 = have_pyqt5()
     72         self._have_backend = have_backend()

/usr/lib/python3.5/site-packages/OCC/Display/backend.py in have_pyqt4()
     35     global HAVE_PYQT4, QtCore, QtGui, QtWidgets, QtOpenGL
     36     try:
---> 37         from PyQt4 import QtCore, QtGui, QtOpenGL
     38         QtWidgets = QtGui
     39         HAVE_PYQT4 = True

RuntimeError: the PyQt4.QtCore and PyQt5.QtCore modules both wrap the QObject class
```

jf---, 2016-04-06:
@tpaviot , indeed, the above demonstrates this perfectly...

handling abstraction of PySide, PyQt4 & PyQt5 is not that trivial.
perhaps we should rather rely on [qtpy](https://github.com/spyder-ide/qtpy/tree/master/qtpy) to handle the abstraction? these days with conda & pip, Yet Another Dependency isnt such a big deal. Your 2 cents @tpaviot ?

tpaviot, 2016-04-06:
@jf--- I'm quite sure you can find a quick fix without any additional dependency

jf---, 2016-04-06:
@tpaviot , well, `backend.py` is the exact same scope as `qtpy`, but done quick & dirty, so doing it the `qtpy` way is cleaner. Also, that code is not completely trivial and accounts for quite a nr of edge cases.

that said, this was a pretty braindead bug, see PR
thanks again @cristobaltapia for reporting...

tpaviot, 2016-04-07:
@cristobaltapia Does the PR #235 fix the problem ?

jf---, 2016-04-08:
hi @tpaviot, I've created a conda env with pyqt4 & pyqt5 to reproduce the problem... so yep, works...

cristobaltapia, 2016-04-09:
Hi. I also tested the fix and it works. Thanks for the fix.

jf---, 2016-04-09:
awesome, thanks for the feedback @cristobaltapia

PR #235 (2016-04-13): fixes #225
fixes #225
commit 64913409 (2016-04-13): fixes #225
commit a974fd49 (2016-04-13): Merge pull request #235 from jf---/jf/225

fixes #225
commit 75190084 (2016-04-13): fixes #225, pyqt4 and 5 conflict
closed 2016-04-13 by tpaviot