tpaviot/pythonocc-core #228: Is it possible to do simple reverse analysis for 3D models with PythonOCC?

huskier, 2016-03-07:
Take a section bar example, I want to get the cross-section profile (or wire) of the section bar, and then extrude from the wire to get the length I want. It‘s a kind of reverse engineering for digital 3D Model.

Further more, I want to get the cross-section profile (or wire) saved in a file for further application.

Could I do the above mentioned analysis with PythonOCC?

jf---, 2016-03-07:
hi @huskier ,

That's an interesting idea... 
Why don't you try to:

- loop through the wires of your solid ( -> cross wire profile )
- are there assumptions you can make of the cross wire profile ( eg, wire has 4 edges between 20 and 60mm, or has 12 edges, or the largest nr of edges over a small area? ), such to identify it?
- if you can identify it, just looping through the [Topology](https://github.com/tpaviot/pythonocc-utils/blob/master/OCCUtils/Topology.py) will allow you to extract it.

huskier, 2016-03-08:
@jf--- Thanks for the help.
Since I am not familiar with PythonOCC, could you please give me a working example for the model https://github.com/huskier/images/blob/master/SectionBar.step？

jf---, 2016-03-08:
hi @huskier ,

Sorry, that sounds like homework :wink: 
This would be a great way for you to get your hands dirty with pythonocc

jf---, 2016-03-29:
@tpaviot please close

closed 2016-03-29 by tpaviot