tpaviot/pythonocc-core #229: RuntimeError: Standard_OutOfMemory NCollection_IncAllocator: out of memory

chenkianwee, 2016-03-23:
Hi I keep running into this memory problem while using pythonocc. I am doing multiple loops (thousands of them), in each loop i boolean_common 2 solids to test if they are touching or intersecting each other. If there is an intersection the two solids are touching. So my question is there a way to free up the memory in pythonocc, or alternatively is there another way of trying to find out if two solids are intersecting or touching each other that is less memory intensive. Thanks!

[image]

tpaviot, 2016-03-23:
Memory leaks have been recently fixed, but the result is not released yet in a new pythonocc version. Adam Lange already reported a success testing his application using the new memory management algorithm (see https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/pythonocc/JtgGUO6nvmQ/hfard_6EDAAJ), please download/compile/install latest pythonocc-core master (from https://github.com/tpaviot/pythonocc-core) and send us your feedback.

chenkianwee, 2016-03-24:
great I will do that, see if it works and provide feedbacks. Thanks.

jf---, 2016-03-29:
@chenkianwee , can you provide us with feedback or a script to replicate the issue?

jf---, 2016-03-31:
@tpaviot , lets close this issue?

chenkianwee, 2016-04-04:
Hi Sorry, for the late reply, was sick for the whole week. Its quite a long script, I think it will be quite hard to replicate the problem with a short script. (it works fine when dealing with a small amount of data). I still do not have the time to recompile a new pythonocc and test out the problem. I guess you can close the issue first, I will get my hands on the issue again in the next week or so and will provide feedback to you guys then. Thanks for the help.

closed 2016-04-04 by chenkianwee

jf---, 2016-04-04:
@chenkianwee , thanks for now, looking fwd having a script if you manage to reproduce ( could be solved... fingers X'd :sunglasses: )