tpaviot/pythonocc-core #230: CSF_GraphicShr is not properly set on conda/windows

tpaviot, 2016-03-29:
see this thread https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/pythonocc/z-NI9A-Vyic/8QjD81q-AwAJ

jf---, 2016-03-29:
hi @tpaviot ,

Could this be compiler specific? I've installed @rainman110 windows packages, that worked out of the box. Is it possible that this plays a factor?

rainman110, 2016-03-29:
The only compiler specific thing is the name of the TKOpenGl library. With MSVC, the name is TKOpenGl.dll, with MinGW  it is libTKOpenGl.dll.

In OCCViewer, we set the CSF_GraphicShr variable to "TKOpenGL.dll", if it is not already set. This would make problems with mingw builds. Since we don't ship these builds, this is probably not the reason for the issue.

Another potential cause of an error is, that the CSF_GraphicShr  variable is already set in the system (e.g. by an opencascade install, or maybe an old install of pythonocc?).

tpaviot, 2016-04-05:
ok, I close the issue

closed 2016-04-05 by tpaviot