tpaviot/pythonocc-core #231: _STEPCAFControl.so wont build

jf---, 2016-03-29:
running into an issue building the current HEAD:

```bash
[ 77%] Swig source
[ 77%] Linking CXX shared module Darwin/x86_64-MinSizeRel-64/_STEPCAFControl.so
Undefined symbols for architecture x86_64:
  "MoniTool_DataMapOfShapeTransient::Clear()", referenced from:
      _wrap_delete_STEPCAFControl_Writer(_object*, _object*) in STEPCAFControlPYTHON_wrap.cxx.o
  "vtable for XSControl_Reader", referenced from:
      XSControl_Reader::~XSControl_Reader() in STEPCAFControlPYTHON_wrap.cxx.o
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [Darwin/x86_64-MinSizeRel-64/_STEPCAFControl.so] Error 1
make[1]: *** [CMakeFiles/_STEPCAFControl.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

could this be a compiler ( clang on osx ) specific issue?

tpaviot, 2016-03-29:
According to travis, no compilation issue is raised on osx and linux with current master

jf---, 2016-03-29:
true, though the osx image on travis 

travis:
```bash
$ clang --version
Apple LLVM version 6.0 (clang-600.0.54) (based on LLVM 3.5svn)
Target: x86_64-apple-darwin13.4.0
```

my local machine:

```bash
$ clang --version
Apple LLVM version 7.0.2 (clang-700.1.81)
Target: x86_64-apple-darwin14.5.0
Thread model: posix
```

clang is becoming increasingly pedantic, for instance, see @rainman110 recent commit: "Workaround clang optimizations for null references"

hence my paranoia ;)

jf---, 2016-03-29:
Ah! Sorry @tpaviot , this is wrt [review/tmp-port-to-oce-0.17.1](https://github.com/tpaviot/pythonocc-core/commits/review/tmp-port-to-oce-0.17.1). Strangely enough, I've been compiling / working with this branch pretty intensively, hence my surprise not being able to compile it.

closed 2016-04-08 by jf---