tpaviot/pythonocc-core #239: Missing topology class

jf---, 2016-04-08:
IMHO (re)moving the Topology class to [OCCUtils](https://github.com/tpaviot/pythonocc-utils/blob/master/OCCUtils/Topology.py) has been a serious mistake...
I'd like to propose to undo this

- the module is well tested and has been part of the project since day 1
- its almost impossible to do anything interesting without it
- 

```python
    afillet = BRepFilletAPI_MakeFillet(cube)
    cnt = 0
    # find edges from vertex
    _map = TopTools_IndexedDataMapOfShapeListOfShape()
    topexp_MapShapesAndAncestors(cube, TopAbs_VERTEX, TopAbs_EDGE, _map)
    results = _map.FindFromKey(vert)
    topology_iterator = TopTools_ListIteratorOfListOfShape(results)
    while topology_iterator.More():
        edge = topods_Edge(topology_iterator.Value())
        topology_iterator.Next()
```

should be:

```
    afillet = BRepFilletAPI_MakeFillet(cube)
    edges = Topology(afillet).edges_from_vertex(vert)
```

Its __really__ annoying coming up with interesting examples without this module.

for example, OCE 0.17 has a AIS_ColoredShape.
this lets you assign a color to subshapes of a TopoDS_Shape
so, something that would be a no-brainer now is a pain in the ass because of the lack of Topology.py, and since we have a proper module for topology traversal, i feel that this is artificially complex.

Missing out on the module violates the `pythonic` concept in many levels:

- "There should be one-- and preferably only one --obvious way to do it."
  - traversing topology should be handled through Topology.py

- "If the implementation is hard to explain, it's a bad idea"
   - the code above makes my eyes bleed... we've worked hard on Topology.py such to never have to deal with this mess again ;)

- "Although practicality beats purity"
   - I do appreciate the division between pythonocc related modules ( pythonocc-utils / pythonocc-contrib ), but factoring out Topology.py has been a bad decision IMO. Perhaps at the time of the "great refactor" project this could be argued for, but now it comes across as highly impractical.

tpaviot, 2016-04-13:
@jf--- This has already been discussed. The topology class won't be restored back in the occ/Utils.

However, it can be added as an example (for instance core_topology_traverse.py), fully or a small part of it. If ever some other example do require this class, it can be imported from core_topology_traverse.

closed 2016-04-13 by tpaviot

jf---, 2016-04-13:
that would be acceptable, but there is no file `core_topology_traverse.py` ( well, not in master ):

[image]

tpaviot, 2016-04-13:
true, but it has to be added

jf---, 2016-04-13:
Sorry, I misread your comment.
Yes, lets do that, its an OK compromise!