tpaviot/pythonocc-core #242: OpenGL VSync / no double buffering

jf---, 2016-04-08:
recently, more OpenGL settings can be set when creating a viewer #238 ,
OCC 7.0.0 has a proper flag [to control this](http://dev.opencascade.org/doc/refman/html/class_open_gl___caps.html#af0b8885a874245deddde60357d7ba5f6), ``swapInterval``
```
Standard_Integer OpenGl_Caps::swapInterval
controls swap interval - 0 for VSync off and 1 for VSync on, 1 by default
```

Since VSync is now off, a lot of [tearing](http://stackoverflow.com/questions/2656531/tearing-in-opengl) is happening ( eg, no double buffering ). This has terrible effects on rendering quality. 

my suggestion is to add another flag ( API consistent then with OCC 7.0.0 ) to Display::Init and find out how this can be handled when creating the viewer.

jf---, 2016-04-08:
in former times, this could be set in OCCViewer:

```python
    def SetDoubleBuffer(self, on_or_off):
        """enables double buffering when shapes are moved in the viewer
        a very shaky picture is drawn, since double buffering
        is disabled by default. fixed here..."""
        self.View.Window().GetObject().SetDoubleBuffer(bool(on_or_off))
```

this API is invalid since OCC 6.8.0

jf---, 2016-04-08:
might be as simple as SetImmediateMode(False)

tpaviot, 2016-04-28:
@jf--- After reading the current oce master, the SetImmediateMode method does not exist. It should be SetImmediateModeDrawToFront

jf---, 2016-05-24:
@tpaviot , that worked out OK... thanks...

closed 2016-05-24 by jf---