tpaviot/pythonocc-core #244: automatic test of examples

jf---, 2016-04-08:
we've been thinking about using the examples as unit tests and to be able to run these automatically.

- to create test from the examples is easy, just add assert statement
- to bypass running the GUI, I suggest using a decorator for the functions to be tested
  - the decorator sets os.setenv("TEST_PYOCC", 1)
  - now, when loading `from OCC.Display.SimpleGui import init_display` we just return a mock object instead of a OCC.Display instance. Any method called simple wont perform any actions

test coverage of our examples would be a great thing to have, especially when working towards a new OCE release.

tpaviot, 2016-04-12:
test at https://github.com/tpaviot/pythonocc-core/tree/review/run-examples-as-tests

jf---, 2016-04-12:
Cool! 
Happy to hear you think its worthwhile to further explore...
I'll start refactoring & placing __assert__ statement ( "assert SomeOperation.IsDone() ) for example

tpaviot, 2016-04-27:
Done. See #258

closed 2016-04-27 by tpaviot