tpaviot/pythonocc-core #245: missing example: render a surface with its gaussian curvature

jf---, 2016-04-08:


tpaviot, 2016-05-03:
@jf--- Can you develop further "it seems that AIS_InteractiveObject cannot be inherited from" ? Do you have an example ?

jf---, 2016-05-03:
Sure!
```python
from OCC.AIS import AIS_InteractiveObject

class AIS_Mesh(AIS_InteractiveObject):
    def __init__(self):
        super(AIS_Mesh, self).__init__()

aaa = AIS_Mesh()
Traceback (most recent call last):
  File "/Users/jelleferinga/miniconda/envs/PyRAPID_QML/lib/python2.7/site-packages/IPython/core/interactiveshell.py", line 3066, in run_code
    exec(code_obj, self.user_global_ns, self.user_ns)
  File "<ipython-input-5-e96e7b6e129b>", line 1, in <module>
    aaa = AIS_Mesh()
  File "<ipython-input-4-e66f40a435a0>", line 3, in __init__
    super(AIS_Mesh, self).__init__()
  File "/Users/jelleferinga/miniconda/envs/PyRAPID_QML/lib/python2.7/site-packages/OCC/AIS.py", line 7150, in __init__
    raise AttributeError("No constructor defined")
AttributeError: No constructor defined
```

Also I recall the .GetHandle method being problematic.

A bit of context:

- Do you remember the gaussian curvature example of back in the day? Alexander Peña made a more idiomatic implementation using [`AIS_InteractiveObject`](https://github.com/jf---/openshapefactory/blob/dc912c5067b66df881297d154f5dbc1b18d1d58e/SFMQTDLL/src/src/AIS_Gauss.cxx) which I'd like to port to pythonocc
- Imagine being able to render the [Frenet frames](https://www.google.nl/search?q=frenet+frame&espv=2&biw=1745&bih=895&source=lnms&tbm=isch&sa=X&ved=0ahUKEwjvqPKArL3MAhUECcAKHa0cCy0Q_AUIBigB#imgrc=tvV-ppmgfQwt_M%3A) of an edge when selecting it 
- I've made a fast mesh importer ( used to abuse SMESH for this ) through the assimp lib which supports a [huge nr of mesh formats](http://www.assimp.org/main_features_formats.html), so an AIS_assimp(the_mesh_file) would be cool. I'm using Prs3d_Presentation for now.

Short and sweet, AIS_InteractiveObject is the gateway to support interesting rendering features ;)

---

I'll look into StdPrs, looks like interesting things can be achieved...

tpaviot, 2016-05-03:
The AIS_InteractiveObject class is an abstract c++ class, thus not define any constructor. The constructor is defined at the subclass level (AIS_Shape fo instance). As a consequence, AIS_InteractiveObject cannot be instanciated. Maybe a new issue should be opened related to this subclassing issue

jf---, 2016-05-03:
Yes, so that's precisely the issue I ran into... 
Ah, I remember it more precisely; so if you don't implement `__init__()` you can't use the `.GetHandle()` method... 

I've renamed the issue...

jf---, 2016-05-05:
hi @tpaviot ,

So just to illustrate the idea; 

Here's an example where rendering of an edge / spline is augmented by rendering the Frenet Frames at a given interval. Its interesting to see the inflection points for instance. Can you imagine that this rendering style is invoked when selecting an edge for example?

So, now, this is implemented, in the same way at the fast_opengl_rendering example.
Its quite different ( well, the same but a lot lower level ) from the standard `display.Context.Display(xxx)` idiom. The reason why `AIS_InteractiveObject` is so interesting that its lets you implement interesting ways of rendering, well integrated in the visualization pipeline.

Next, when we'll have the GLSL shader working, then its pretty unlimited of what we can achieve in terms of rendering, while pretty easy to implement 😎 

[image]