tpaviot/pythonocc-core #247: TypeError: in method 'Display3d_Init', argument 2 of type 'long'

cristobaltapia, 2016-04-09:
Me again. The problem with qt5 is solved, but now I have this one:

```
from OCC.Display.SimpleGui import init_display
from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox

display, start_display, add_menu, add_function_to_menu = init_display()

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-1-a54418bdea70> in <module>()
      2 from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox
      3 
----> 4 display, start_display, add_menu, add_function_to_menu = init_display()
      5 

/usr/lib/python3.5/site-packages/OCC/Display/SimpleGui.py in init_display(backend_str, size)
    142         win = MainWindow()
    143         win.show()
--> 144         win.canva.InitDriver()
    145         display = win.canva._display
    146         if sys.platform != "linux2":

/usr/lib/python3.5/site-packages/OCC/Display/qtDisplay.py in InitDriver(self)
    121     def InitDriver(self):
    122         self._display = OCCViewer.Viewer3d(self.GetHandle())
--> 123         self._display.Create()
    124         # background gradient
    125         self._display.set_bg_gradient_color(206, 215, 222, 128, 128, 128)

/usr/lib/python3.5/site-packages/OCC/Display/OCCViewer.py in Create(self, create_default_lights)
    133 
    134     def Create(self, create_default_lights=True):
--> 135         self.Init(self._window_handle)
    136         self.Context_handle = self.GetContext()
    137         self.Viewer_handle = self.GetViewer()

TypeError: in method 'Display3d_Init', argument 2 of type 'long'
```
... a qt5 window is opened at least.

jf---, 2016-04-09:
@cristobaltapia , are you on windows?

cristobaltapia, 2016-04-09:
No, I'm using ArchLinux.

tpaviot, 2016-04-23:
Fixed by PR #254

tpaviot, 2016-04-24:
Fixed by PR #254

closed 2016-04-24 by tpaviot

davtalab, 2017-05-31:
Hi,

I also get this error:

TypeError: in method 'Display3d_Init', argument 2 of type 'long'

could not fix it yet. Can you please help on this?