tpaviot/pythonocc-core #252: compiling pythonocc 0.16.4

chenkianwee, 2016-04-18:
Hi I have been trying to compile 0.16.4 so that I can test out new memory management system. However, its just so difficult to do a compilation for a not so experience programmer like me. The conda packages are not updated to the 0.16.4. So any help will be appreciated. I have attached a print screen of the error I faced. I cmake the pythonocc src for mingw-64 compiler. It just totally do not compile at all. 
[image]

tpaviot, 2016-04-20:
I have the feeling it's related to the swig version you use. I've been reported that swig 3.0.7 fails, maybe it's the same for 3.0.8, let me test that. In the meantime, you can test with an older swig version, e.g. 3.0.1 (the one I'm using)

chenkianwee, 2016-04-20:
I tried with swig 3.01 and it still did not work. I will try again and see if I can compile it with other versions.

[image]

jf---, 2016-04-20:
@chenkianwee , an alternative is to grab the [conda binaries](https://anaconda.org/DLR-SC/pythonocc-core/files)

tpaviot, 2016-04-20:
The thing is that 0.16.4 is not available on dlr-sc. I dis ont make an
announcement for it, my fault

chenkianwee, 2016-04-21:
I have tried older versions of swig and I am still having the same error msg. I am using the swig.exe in the swigwin package. I read its for 32 bit windows, do you think that could be the problem ?

C:\pythonocc-core-0.16.4\cmake-build>mingw32-make
mingw32-make[2]: *** No rule to make target 'D:/Devel/oce-win-bundle/dist64/bin/
libfreetype.dll', needed by 'win64/bin/MinSizeRel/_V3d.pyd'.  Stop.
CMakeFiles\Makefile2:66: recipe for target 'CMakeFiles/_V3d.dir/all' failed
mingw32-make[1]: *** [CMakeFiles/_V3d.dir/all] Error 2
Makefile:148: recipe for target 'all' failed
mingw32-make: *** [all] Error 2

it is not able to find the libfreetype.dll ? In a directory that is totally not on my com ?

tpaviot, 2016-04-23:
I don't understand. Are you trying to build oce or pythonocc ? both ? It seems you're not familiar with compiling with cmake and a c++ compiler. I suggest you wait for the binaries to be available on conda before testing. I can't see any bug here (I tested with different versions of swig on my machine, everything is ok), let me close this issue.

closed 2016-04-23 by tpaviot