tpaviot/pythonocc-core #253: init_display method does not work

kozintsev, 2016-04-19:
Hello All. 
I test quick example:

`
from OCC.Display.SimpleGui import init_display
from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox

display, start_display, add_menu, add_function_to_menu = init_display()
my_box = BRepPrimAPI_MakeBox(10., 20., 30.).Shape()

display.DisplayShape(my_box, update=True)
start_display()
`

And I have problem:

C:\Anaconda2\python.exe C:/Users/Oleg/PycharmProjects/untitled2/main.py
INFO:OCC.Display.backend:backend loaded: qt-pyqt4
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt4
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Traceback (most recent call last):
  File "C:/Users/Oleg/PycharmProjects/untitled2/main.py", line 4, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "C:\Anaconda2\lib\site-packages\OCC\Display\SimpleGui.py", line 143, in init_display
    win.canva.InitDriver()
  File "C:\Anaconda2\lib\site-packages\OCC\Display\qtDisplay.py", line 123, in InitDriver
    self._display.Create()
  File "C:\Anaconda2\lib\site-packages\OCC\Display\OCCViewer.py", line 118, in Create
    self.Init(self._window_handle)
RuntimeError: Aspect_DriverDefinitionError
�� ������ ��������� ������.

Process finished with exit code 1

My computer:
Windows 7 Pro 64-bit
Anaconda2-4.0.0-Windows-x86.exe

And I install pythonocc use command:
conda install -c https://conda.anaconda.org/dlr-sc pythonocc-core

jf---, 2016-04-19:
hi @kozintsev 

this is unusual... 
lets see, my guess is that your OpenGL drivers might be outdated...
Is it possible that you have an Intel HD Graphics card?
IIRC, there's been a case where updating the driver resolved the issue...

thanks for the detailed feedback so far...

kozintsev, 2016-04-19:
Yes, My computer has 2 video card:
- Intel HD Graphics
- AMD Radeon HD 6650M [Lenovo]

I tried to update driver but I got error:

INFO:OCC.Display.backend:backend loaded: qt-pyqt4
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt4
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Traceback (most recent call last):
  File "c:\Anaconda3\hello.py", line 4, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "C:\Anaconda3\lib\site-packages\OCC\Display\SimpleGui.py", line 143, in init_display
    win.canva.InitDriver()
  File "C:\Anaconda3\lib\site-packages\OCC\Display\qtDisplay.py", line 123, in InitDriver
    self._display.Create()
  File "C:\Anaconda3\lib\site-packages\OCC\Display\OCCViewer.py", line 118, in Create
    self.Init(self._window_handle)
RuntimeError

My new configuration:
Windows 7 Pro 64-bit
Anaconda3-4.0.0-Windows-x86_64.exe

And I installed pythonocc use command:
conda install -c https://conda.anaconda.org/dlr-sc pythonocc-core

Thank you for attention!

jf---, 2016-04-19:
Same thing huh... 
So on mac, there's a little utility that lets you select the intel HD card or the nvidia.
Is there something similar for your machine?
In other words; it would be interesting to know whether the issue is similar for one card or the other...

kozintsev, 2016-04-19:
Is there something similar for your machine?
Yes, Radeon is chosen for python.exe.
The error is the same.
On my computer works, for example xBIM Xplorer (use OpenCASECADE), and worked old version pythonocc.
I do not remember exactly what.

kozintsev, 2016-04-19:
I tried to run old version PythonOCC (0.5)
And...

C:\Python26>python.exe helloworld.py
C:\Python26\lib\site-packages\OCC\Display\SimpleGui.py:116: wxPyDeprecationWarni
ng: Using deprecated class PySimpleApp.
  app = wx.PySimpleApp()
C:\Python26\lib\site-packages\OCC\Display\SimpleGui.py:117: wxPyDeprecationWarni
ng: Call to deprecated item.
  wx.InitAllImageHandlers()
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
WNT Graphic device created.
WNT window created.
Viewer created.
*OpenGL interface: SetPixelFormat failed. Error code 2000
Traceback (most recent call last):
  File "helloworld.py", line 21, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "C:\Python26\lib\site-packages\OCC\Display\SimpleGui.py", line 121, in in
it_display
    win.canva.InitDriver()
  File "C:\Python26\lib\site-packages\OCC\Display\wxDisplay.py", line 145, in In
itDriver
    self._display.Create()
  File "C:\Python26\lib\site-packages\OCC\Display\OCCViewer.py", line 110, in Cr
eate
    self.Init(self._window_handle)
RuntimeError: Visual3d_ViewDefinitionError
Association failed

Probably I have problem with my OS.

kozintsev, 2016-04-19:
It's work!

Python 2.7.11 
QT4
PythonOCC 0.16.3

Probably, driver's problem.
Thank's!

closed 2016-04-19 by kozintsev

jf---, 2016-04-19:
@kozintsev , would you be so kind as to check this [previous duplicate issue](https://github.com/tpaviot/pythonocc-core/issues/170)

jf---, 2016-04-19:
@kozintsev , awesome... 
I can recommend using the [conda binaries](https://anaconda.org/DLR-SC/pythonocc-core/files)