tpaviot/pythonocc-core #257: gp_Vec can't be divided by float

tpaviot, 2016-04-26:
On python2/pythonocc-0.16.4
```
>>> from OCC.gp import gp_Vec
>>> gp_Vec(1,2,3)/2
<OCC.gp.gp_Vec; proxy of <Swig Object of type 'gp_Vec *' at 0x02B3A728> >
````

On python3:
```
>>> from OCC.gp import gp_Vec
>>> gp_Vec(1,2,3)/2
>>>Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported operand type(s) for /: 'gp_Vec' and 'int'
```
why ?

tpaviot, 2016-04-26:
Note: on python3, other operators work (+,-,*).

tpaviot, 2016-04-29:
This issue is related to SWIG-3*. Using SWIG-2*, it works as expected

rainman110, 2016-05-02:
Could it be something with float vs integer division? Try:
```
gp_Vec(1,2,3)/2.
```
(with the dot after the 2)

tpaviot, 2016-05-02:
It seems to be a SWIG 3 issue (see https://github.com/swig/swig/pull/617 and https://github.com/swig/swig/pull/136)

tpaviot, 2016-05-02:
Note that a workaround is to replace
```
a_gp_Vec / a_float
```
with
```
a_gp_Vec * ( 1. / a_float)
```
same for integers

rainman110, 2016-05-25:
@tpaviot Should we then stick to SWIG 2 for the packaging?

tpaviot, 2016-05-25:
@rainman110 If possible yes

rainman110, 2016-05-25:
@tpaviot  I just tested it with swig 2.0.10. There, I still get the error on python3:
```TypeError: unsupported operand type(s) for /: 'gp_Vec' and 'int'```
or
```TypeError: unsupported operand type(s) for /: 'gp_Vec' and 'float'```

rainman110, 2016-06-01:
@tpaviot I just tested this issue with the newest swig 3.0.9. This fixes the issue.

What remains to be done is to rebuild all conda packages with the 3.0.9 version of swig.

jf---, 2016-06-01:
nice find @rainman110 !

rainman110, 2016-06-01:
I uploaded all python3 related packages.

@tpaviot The issue can be closed.

jf---, 2016-06-01:
is there a unittest for gp_Vec division?

tpaviot, 2016-06-04:
I added a unittest (see #278). @rainman110 I'll close the issue as soon as the test passes on travis.

tpaviot, 2016-06-04:
Fixed. Thanks you @rainman110

closed 2016-06-04 by tpaviot