tpaviot/pythonocc-core #259: Standard errors like Standard_OutOfRange not caught

jf---, 2016-04-28:
For example:

```python
from OCC.gp import *
d = gp_Dir()
d.Coord(-1) # 1,2 or 3 are valid indices
# KABOOM
libc++abi.dylib: terminating with uncaught exception of type Standard_OutOfRange
```

Could this possible be compiler related?

tpaviot, 2016-04-28:
Same issue on windows/mingw, it's not compiler specific

tpaviot, 2016-04-28:
It used to work, I can't understant when it started to fail, and why. A quick check doesnt't show any obvious clue. I don't know if something changed in occ exception handling

jf---, 2016-04-28:
hi @tpaviot , yes, I recall on earlier version that pyocc wrapped these as RuntimeErrors...
So possible, this is a meaningful future unittest?

this comes up at the [forum](http://www.opencascade.com/content/how-catch-exceptions):

> You might want to check your compile flags. For some operations, CASCADE does not really throw a proper exception and performs as you describe. You must compile without the -DNO_CXX_EXCEPTION flag. If you have that flag set, CASCADE will abort instead of throwing the exception. This took us several days to find.

tpaviot, 2016-04-28:
Seems related to #13.

Everything is compiled with the right flag OCC_CONVERT_SIGNALS, the try/catch is done as the official doc stands. I suspect a conflict with builtin siwg exception, I'll check that

jf---, 2016-04-28:
Shall we close either issue, and mark it duplicate?
Right, wasnt this around the time that we've moved from SWIG 2 -> 3?

tpaviot, 2016-04-29:
* this is not related to the SWIG version (I tried with both 2 and 3, same issue)

* it's not related to the builtin SWIG exception handling system (tested with and without, same issue).

jf---, 2016-04-29:
Thanks for the heads up... that's quite a nr of possibilities you've excluded...
Hmm, so your suspicion is directed towards OCE?

jf---, 2016-04-29:
the other day @rainman110 offered to fire up valgrind for the failing #124 ( osx specific, thanks btw! ). possible this issue might be of greater urgence?

tpaviot, 2016-04-29:
I tried to compile in RelWithDebInfo mode, rather than MinSizeRel, no change.

Not sure about the issue priorites, it depends on so many factors to be fully objective, let's talk about that in another issue if necessary.

rainman110, 2016-04-29:
@tpaviot This error is totally valid. C++ throws an exception by accessing an invalid array index.

PythonOCC does not catch this c++ exception. Hence #KABOOM :grin: 

To fix this, check also for std::exception(s) in the wrapper.

rainman110, 2016-04-29:
What I would first test, is a C++ only implementation. Then try to catch Standard_Error. If this is also not working, then oce does not throw the exceptions properly.

What one should keep in mind is, that the exception mechanism is compiler (version) specific. Thus, one has to make sure, that one uses the same compiler for everything.

jf---, 2016-04-29:
> What one should keep in mind is, that the exception mechanism is compiler (version) specific. Thus, one has to make sure, that one uses the same compiler for everything.

That's most insightful...

tpaviot, 2016-04-30:
I confirm that the Standard_OutOfRange exception is raised by oce, and pythonocc fails to catch it. The exception mechanisme maybe compiler dependant, but I remember it used to work on *all* platforms, and that now it fails also on *all* platforms, whatever the compiler is (gcc, mingw, clang etc.).

rainman110, 2016-05-01:
This is not a problem with exception catching. The catch(Standard_Failure) string stuff is not implemented correctly.

I am working on a fix. I'll make a PR as soon I am finished and the tests are running.

rainman110, 2016-05-01:
All tests run fine, also the d.Coord(-1) example. I'll make a PR.

rainman110, 2016-05-02:
Can we close this issue then?

jf---, 2016-05-02:
@rainman110 sure, thanks so much for your input on this...

closed 2016-05-02 by jf---

jf---, 2016-05-02:
Gents,

I think this is a significant fix, and, like the memory handling refactor, is sufficient ground to carve out a release?

tpaviot, 2016-05-02:
Yes, I finish the fix for #261 first, and I'd also like to understand what happens with #257

jf---, 2016-05-02:
OK, that sounds perfect...
That said, I cant offer much help there, I'll try and put together the MakeVolume example