tpaviot/pythonocc-core #260: cross section visualization example

jf---, 2016-04-28:
[see here](http://dev.opencascade.org/index.php?q=node/839)

[image]

aothms, 2016-04-28:
Wow nifty. Is this a rendering step am I right, stencil buffer magic probably, no boolean ops involved?

jf---, 2016-04-28:
Exactly... I guess with the WIP wrt new IFC gui's this might come in handy ;)
Warning, this might have a dependency on #5

tpaviot, 2016-04-29:
@jf--- where's the source code ? I can't fint it

jf---, 2016-04-29:
hi @tpaviot , so this is a visual reminder to self ( alas, can't "assign" issues  ) to create this example.

tpaviot, 2016-10-12:
@jf--- any progress regarding this issue ?

jf---, 2016-10-12:
@tpaviot not yet, of lesser urgence...

tpaviot, 2017-05-04:
commit c0d6d5f adds an example using clipping plane

tpaviot, 2017-05-04:
see https://youtu.be/qZhkGUvdAZU

closed 2017-05-04 by tpaviot