tpaviot/pythonocc-core #267: Decrease build time

tpaviot, 2016-05-10:
Building pythonocc-core is a long task. Continous Integration services have time limits (for instance travis is 50mn, appveyor 1h), and pythonocc-core builds exceed this limit. I tried to 

So far, I see the following solutions:

* ask travis an extra time. Done I will give here the answer from the travis-ci support

* Prebuild the cxx file on our Windows/Linux/OSX machines and push them to github. Thus skip the SWIG stage when building core. The cxx generation could be achieved by the pythonocc-generator

* use the cache systems of travis and apveyor to avoid SWIG'ing each time a new build is started. (SWIG should only be run whenever .i files are modified, actually this is wht's happening when we build pythonocc-core on our machines).

tpaviot, 2016-05-11:
I just asked an extra time to travis support this morning, they gave 20 more minutes ! That means the time limit is now 70minutes, which is just perfect for building everything needed.

jf---, 2016-05-11:
Wow! That's a lot more convenient that having to deal with ccache ;)
Sweet...

tpaviot, 2016-05-12:
I close the issue, no doubt it will have to be reopened in the midterm

closed 2016-05-12 by tpaviot