tpaviot/pythonocc-core #271: cannot fuse cylinder and sphere

DavidAustin, 2016-05-24:
I'm having a problem fusing a cylinder and a sphere:

from OCC.StlAPI import StlAPI_Writer
import OCC.BRepPrimAPI
import OCC.BRepAlgo

f = OCC.BRepAlgo.BRepAlgo_Fuse(
    OCC.BRepPrimAPI.BRepPrimAPI_MakeCylinder(6.0, 100).Shape(),
    OCC.BRepPrimAPI.BRepPrimAPI_MakeSphere(6.0).Shape()
).Shape()

stl_export = StlAPI_Writer()
stl_export.Write(f, 'fuse_prob.stl', True)

produces an object with a hemisphere and a flat disc - i.e. missing the 'body'
of the cylinder.  I'm using pythonocc v 0.16.4

Same problem with writing a STEP file and the geometry seems wrong too - 
distance testing seems to suggest a hole in the middle of the object.

Thanks,
David

tpaviot, 2016-05-24:
@DavidAustin Prefer the ```BRepAlgoAPI_Fuse``` class from the ```OCC.BRepAlgoAPI``` module.

```
from OCC.StlAPI import StlAPI_Writer
import OCC.BRepPrimAPI
import OCC.BRepAlgoAPI

f = OCC.BRepAlgo.BRepAlgoAPI_Fuse(
OCC.BRepPrimAPI.BRepPrimAPI_MakeCylinder(6.0, 100).Shape(),
OCC.BRepPrimAPI.BRepPrimAPI_MakeSphere(6.0).Shape()
).Shape()

stl_export = StlAPI_Writer()
stl_export.Write(f, 'fuse_prob.stl', True)
```
That should work. Please report success/failure.

DavidAustin, 2016-05-29:
Hi Thomas,

Thanks - that fixed it.

David

closed 2016-05-30 by tpaviot