tpaviot/pythonocc-core #274: Image_PixMap, no image library available! Image saved in PPM format.

chenkianwee, 2016-05-24:
Hi I am using pythonocc-core 0.16.5 and I am experiencing this error while trying to save the image from the occviewer. What seems to be the issue? Am I missing some libraries?

jf---, 2016-05-24:
the `core_display_export_to_image.py` example gives a hint:

```pytohn
# in order to be able to export to usual file formats
# oce must have been compiled with FreeImage
# otherwise, images will be exported to PPM format
```

are you using the conda binaries or did you roll your own?

rainman110, 2016-05-24:
Somehow, FreeImage is not linked. I suppose this is my fault.

@chenkianwee  What OS do you use?

I had some issues on windows, because FreeImage was updated by Conda (not by us) but does not ship FreeImagePlus. Therefore, cmake does not find FreeImage completely and OCE is built without FreeImage support.

chenkianwee, 2016-05-24:
@jf--- I installed the conda binaries 
@rainman110  I am using windows 8. 

Ya I suspected it might have to do with the FreeImage library, but I checked I have the library installed. Hmnn so if thats the case, is there a way to solve this ?

rainman110, 2016-05-24:
@chenkianwee No, you can't fix this. I need to fix the conda packaging and upload new binaries.

rainman110, 2016-05-24:
I fixed the issue for python 2.7. So @chenkianwee , if you're using python 2.7, just update oce via
```
conda update oce
```
The python 3.5 packages are still in the build process and will be ready in the evening.

On my local machine, this fixes the issue.

chenkianwee, 2016-05-24:
Hi @rainman110  thanks for the quick response!! I will get it updated tomorrow morning, its quite late over here. Thanks !!

chenkianwee, 2016-05-25:
I just updated the package and everything is working. Thanks !

closed 2016-05-25 by chenkianwee