tpaviot/pythonocc-core #277: Do you test windows 64 version?

kozintsev, 2016-06-04:
Hello All!

When I run example Hello dumb box an error occurs:

C:\Anaconda3\python.exe C:/Users/Oleg/PycharmProjects/untitled1/main.py
INFO:OCC.Display.backend:backend loaded: qt-pyqt4
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt4
Traceback (most recent call last):
  File "C:/Users/Oleg/PycharmProjects/untitled1/main.py", line 4, in <module>
    display, start_display, add_menu, add_function_to_menu = init_display()
  File "C:\Anaconda3\lib\site-packages\OCC\Display\SimpleGui.py", line 201, in init_display
    win.canva.InitDriver()
  File "C:\Anaconda3\lib\site-packages\OCC\Display\qtDisplay.py", line 109, in InitDriver
 ###### 3D rendering pipe initialisation #####
    self._display.Create()
  File "C:\Anaconda3\lib\site-packages\OCC\Display\OCCViewer.py", line 135, in Create
Display3d class initialization starting ...
    self.Init(self._window_handle)
Aspect_DisplayConnection created.
RuntimeError

Process finished with exit code 1

Windows 7 64-bit, Anaconda 64-bit.

tpaviot, 2016-06-04:
I do think it is related to https://github.com/tpaviot/pythonocc-core/issues/230

Can you please check that the CSF_GraphicShr en var points to the right file ?

kozintsev, 2016-06-04:
Yes, You right! Thank you!

closed 2016-06-04 by kozintsev