tpaviot/pythonocc-core #279: dimensions with PythonOCC

edlala, 2016-06-06:
The dimensions example that is packaged with PythonOCC produces the following error related to units. Tested with version of PythonOCC 0.16.3. 

Traceback (most recent call last):
  File "C:\Users\Emad\Documents\Aptana Studio 3 Workspace1\General\core_dimensions.py", line 39, in <module>
    display.Context.Display(rd.GetHandle())
RuntimeError: Units_NoSuchType
BAD LEXICON descriptor

The problem seems to be related to hard-coding the units path. 
D:\OCE-0.16.1-Mingw32\share\oce-0.16\src\UnitsAPI

A solution would be to allow the user to provide the path.

tpaviot, 2016-06-06:
Use the CSF_UnitsLexicon environment variable to set up the path

edlala, 2016-06-06:
I set up this variable CSF_UnitsLexicon to a path but I still see the
error. See snapshots below.

[image: Inline image 2]

[image: Inline image 1]
[image: Inline image 3]

jf---, 2016-06-06:
@edlala cant see the images... 
does sound like this resembles #139 though

jf---, 2016-09-28:
@tpaviot , this old and kind of overlaps with #139 , shall we close it?

tpaviot, 2016-09-29:
@jf--- It's not the same than #139. I already faced this issue in the past, it's related with installation and path. @edlala did you test/install the last release ? is this issue still open or can it be closed ?

edlala, 2016-09-29:
Hi Thomas,

No, I haven't tested it in the new version, even though I'm continuing to
do cool stuff/application with PythonOCC and wxPython :)

Keep it up!

Thanks,
Emad

jf---, 2016-09-29:
hi @edlala , can you substantiate the claim of doing cool stuff 🍸 ? ( sounds promising... ) any screenshots you're willing to share / satisfy our curiosity?

jf---, 2016-12-22:
so this issue is OCE 0.17 specific and resolved for OCE 0.18.
@tpaviot please close.

smewp, 2017-02-14:
I ran into this same traceback today using pythonocc-core=0.17.2 on a fresh install using an anaconda3 python3.5.1 on windows 10.

edlala, 2017-03-08:
Hi Thomas,

This error still occurs in the newer versions too. It's now even worse
because before in v16.1, I added the units text files manually to the
directory referred.

Now on anaconda 64bit, the error occurs and it doesn't even tell the
directory where the units file should be placed. I really need a solution
to this. It's a showstopper.

Tested on today's anaconda build on Windows 7.

RuntimeError: Units_NoSuchType
BAD LEXICON descriptor

You can test it on the example of dimensions. I also setup the environment
variable you referred to me before but no use.

At least for now, as a workaround that worked in 16.1, if you can tell me
the directory where I place the unit files for the anaconda 64 builds. For
the old 16.1 32bit Python version, it's
D:\OCE-0.16.1-Mingw32\share\oce-0.16\src\UnitsAPI

Appreciate the help and good work.

Best regards
Emad

On Thu, Sep 29, 2016 at 8:55 AM Emad Dlala <emad.dlala@gmail.com> wrote:

Hi Thomas,

No, I haven't tested it in the new version, even though I'm continuing to
do cool stuff/application with PythonOCC and wxPython :)

Keep it up!

Thanks,
Emad

edlala, 2017-03-17:
I wanted to tell you that this problem was overcome by defining the
environment variables to the file paths of the missing data files. This fix
was applied on v0.17.2.

[image: Inline image 1]

tpaviot, 2017-05-03:
Fixed by commit 140534c289834320cda0043df64ee130a9c7cb40

closed 2017-05-04 by tpaviot