tpaviot/pythonocc-core #280: Can't run x3dom rendering example

yoeldk, 2016-06-07:
Hello,

I'm running the code of the rendering cylinder example, but the browser (firefox) shows nothing.
I checked it for x3dom rendering support and it does support it, but when running the code it doesn't show the shape.

Thanks in advance for all helpers,
Joel

jf---, 2016-06-07:
whats the output of your console / command terminal?

yoeldk, 2016-06-09:
There's no special output / error message...
It opens a firfox tab with the "correct" background (blue background with the text "python occ x3dom renderer" at the bottom left) but doesn't show the 3d shape at all.

tpaviot, 2016-06-09:
@yoeldk Which browser version is it ? os ? pythonocc ?

A possible way to debug is to enable the Firefox (click Menu->Debugger->WebConsole). After enabling the webconsole, reload the webpage, you should see a log if ever there is an issue. Can you please copy/paste the issue here ?

yoeldk, 2016-06-10:
[image]

yoeldk, 2016-06-10:
Update:
I'm using windows 7 with a firefox version 47.0 as my default browser.
As seen above, firefox does not render the shape but I copy-pasted the url into a chrome browser and there it worked fine..
I would still prefer to have it working on firefox as well.

jf---, 2016-06-14:
thanks for the update @yoeldk , happy to hear you managed to make this work.
can I ask you to close the issue?

tpaviot, 2016-06-14:
@jf--- no the issue should not be closed, @yoeldk wants to run FF47

jf---, 2016-06-14:
ooops...

tpaviot, 2016-06-14:
@yoeldk It seems you run pythonocc-0.16.3 which is outdated. Can you please update to the latest release 0.16.5 and check again

yoeldk, 2016-06-14:
@tpaviot According to the following website I am running the latest software (0.16.3). What is the basis for the claim that it is outdated?
http://www.pythonocc.org/download/

tpaviot, 2016-06-14:
https://github.com/tpaviot/pythonocc-core/releases/tag/0.16.5

yoeldk, 2016-06-14:
With 0.16.5 it works fine in FF, issue can be closed.

Thanks,
Yoel

closed 2016-06-15 by tpaviot