tpaviot/pythonocc-core #281: Exception initializing display: RuntimeError: Standard_NullObject TCollection_AsciiString : parameter 'astring'

aothms, 2016-06-10:
In the light of nothing ventured, nothing gained: I have some users that run into the following issue when opening the pythonOCC display.

~~~
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
WNT window created.
V3d_Viewer created.
V3d_View created
AIS_InteractiveContext created.
Display3d class successfully initialized.
 ########################################
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\OCC\Display\SimpleGui.py", line 164, in in
it_display
    win.canva.InitDriver()
  File "C:\Python27\lib\site-packages\OCC\Display\pyqt4Display.py", line 85, in
InitDriver
    self._display.SetModeShaded()
  File "C:\Python27\lib\site-packages\OCC\Display\OCCViewer.py", line 149, in Se
tModeShaded
    self.View.SetComputedMode(False)
RuntimeError: Standard_NullObject
TCollection_AsciiString : parameter 'astring'
~~~

This error seems to occur on nearly every call pertaining to the viewer / context / view. My guess is that it is not pythonOCC specific (sorry for that, don't think I will get these users to try compiling some C++).

Looking solely at the parameter name I stumble upon this file:
https://github.com/tpaviot/oce/blob/c0cdb53fdf7cc708eb303bec49b18e27edd7c74b/src/Aspect/Aspect.cxx#L92

The only mantis issue I can find slightly resembling this is:
http://tracker.dev.opencascade.org/view.php?id=26611

This issue pertains to the font manager. Considering the native languages of the users running into this problem, it might be the case that they have some exotic fonts.

My question is twofold: Is this somehow a known issue? Does it make sense to suggest these users to fiddle with their system fonts on account of this issue, i.e are fonts even loaded when the viewer is being initialized?

Thanks for your advice.

tpaviot, 2016-06-15:
@aothms Not known issue, I'll try to see what's going on. How should I reproduce the error ?

tpaviot, 2016-06-17:
@aothms According to Mantis it is fixed in occt691. Patch here http://git.dev.opencascade.org/gitweb/?p=occt.git;a=blobdiff;f=src/Font/Font_FontMgr.cxx;h=583281121b8b77c4dc857091a662c6b6279eff71;hb=508643cf1b5cf05f6bf651c1bbf7e5866e1874fa;hpb=09749bc31cc89ab97c56658e78c1f46049ec86c3

aothms, 2016-06-20:
Yep. My guess is that it is the same issue. I have received a dump of all the fonts of this particular user, and if I install one particular font with filename `¢­Ã++=À--+.TTF` and display name `金桥简仿宋 Medium` I can reproduce the issue. Deleting the font resolves the issue, so there is a clear workaround. Let's just assume the issue is fixed upstream. I will close it.

closed 2016-06-20 by aothms

jf---, 2016-06-20:
Well, until this is resolved we now know the error, and it would be little work to inform a user of what's causing the problem.

aothms, 2016-06-20:
Well, the error I get now is extremely generic: `TCollection_AsciiString : parameter 'astring'` (see top post). I tried a bit to raise this error in another way, but I cannot get closer to this `Font_FontMgr` then by just calling the `InitDriver()` function. So when to inform of what is still a bit unclear to me.