tpaviot/pythonocc-core #282: Error when running the examples in the tree

yoeldk, 2016-06-17:
Hello everyone,

I'm running windows 7 + pyCharm.
I git-cloned this tree using pyCharm's plugin, but unfortunately I get a python error window "python has stopped working" when trying to the run the examples in the tree. Clicking the "view details" button shows the following files:

Files that help describe the problem:
  C:\Users\TEMP\AppData\Local\Temp\WER7E18.tmp.WERInternalMetadata.xml
  C:\Users\TEMP\AppData\Local\Temp\WER89DC.tmp.appcompat.txt
  C:\Users\TEMP\AppData\Local\Temp\WER8B06.tmp.mdmp

Following the log from pyCharm's console:
C:\Users\yoelk\Miniconda3\python.exe C:/pythonocc-core/examples/core_geometry_airfoil.py
INFO:OCC.Display.backend:backend loaded: qt-pyqt4
INFO:OCC.Display.SimpleGui:GUI backend set to: qt-pyqt4
 ###### 3D rendering pipe initialisation #####
Display3d class initialization starting ...
Aspect_DisplayConnection created.
Graphic_Driver created.
WNT window created.

Can anyone advice?

Thanks,
Yoel

jf---, 2016-06-17:
hi @yoeldk ,

> Maybe I have a setup error and I should have actually be using a different python interpreter?

lets rule out any env matters first?
- setting up an pythonocc env is pretty trivial, has your conda env been created with conda?
- check that in pycharm you're using the correct interpreter, corresponding to your environment

let's get you going ;)

yoeldk, 2016-06-17:
1. I installed miniconda and then from the cmd line executed:
`conda install -c https://conda.anaconda.org/dlr-sc pythonocc-core `

2. I get-cloned the tree using a pyCharm plugin, and chose as interpreter the following file:
C:\Users\yoelk\Miniconda3\python.exe . 

Anything I missed?

yoeldk, 2016-06-17:
See following screenshot - looks like a pyQt4 error according to the title of the window:
[image]

tpaviot, 2016-06-18:
@yoeldk Does the example work outside PyCharm ?

yoeldk, 2016-06-20:
@tpaviot  No. I tried through the windows cmd line and it failed with the same error.

yoeldk, 2016-07-05:
I upgraded my laptop and installed everything from scratch (this time on windows 10) and now the example work fine. From my perspective this issue can be closed.

closed 2016-07-06 by tpaviot