tpaviot/pythonocc-core #283: Is there an older version of pre-compiled installer?

sonnyhu, 2016-06-22:
Hi, I am wondering on https://conda.anaconda.org/dlr-sc, is there a way to find older version of pre-compiled installer? Thanks.

jf---, 2016-06-22:
try `conda install pythonocc==0.16.2`

sonnyhu, 2016-06-22:
Without `-c https://conda.anaconda.org/dlr-sc`?

sonnyhu, 2016-06-22:
Got error:

```
Fetching package metadata: ....
Solving package specifications: .
Error:  Package missing in current osx-64 channels: 
  - pythonocc ==0.16.2
```

jf---, 2016-06-22:
oops, i have `https://conda.anaconda.org/dlr-sc` in my .condarc file ( which is where you can put your channels )

so yes, `conda -c https://conda.anaconda.org/dlr-sc pythonocc==0.16.2`

sonnyhu, 2016-06-22:
Thanks for the help, I am getting the same error though:

```
conda install -c https://conda.anaconda.org/dlr-sc pythonocc==0.16.2
Fetching package metadata: ......
Solving package specifications: .
Error:  Package missing in current osx-64 channels: 
  - pythonocc ==0.16.2

You can search for this package on anaconda.org with

    anaconda search -t conda pythonocc ==0.16.2

You may need to install the anaconda-client command line client with

    conda install anaconda-client
```

jf---, 2016-06-23:
hi @sonnyhu ,

- so 0.16.2 is indeed not available on the DLR repo 
- `conda install -n tmp -c dlr-sc pythonocc-core=0.16.2` is the right command
- that said, why would you want to miss you on recent, important fixes? the API from 0.16.2 -> 0.16.5 has no changes

sonnyhu, 2016-06-23:
Hi, @jf---

We are getting segmentation fault by using newer version. It could be our code is not well written to handle different cases but it is a legacy codebase and we don't really want to spend a lot of time debugging it, so we hope to stick with a working version of dependencies.

jf---, 2016-06-27:
hi @sonnyhu ,

I guess 0.16.2 is not available.
Could you close the issue please, since this is not a bug...

Thanks...

sonnyhu, 2016-06-27:
Sure. Thanks.

closed 2016-06-27 by sonnyhu

jf---, 2016-06-27:
@sonnyhu thanks, if you have more on the segfaults we might be able to offer some more help