tpaviot/pythonocc-core #289: pythonOCC with SMESH?

e-dub, 2016-07-12:
Has anyone got this running? Currently trying to build on Ubuntu 16.04.  

The following I got to build
OCE Version 0.17
SMESH Version 6.6
pythonOCC latest (without SMESH integration)

I also tried unsucessfully with the version found here: https://github.com/adamLange/pythonocc-core

Any ideas for me?

Thanks in advance!

tpaviot, 2016-07-14:
@e-dub I've currently being working on that topic.  smesh needs to be upgraded to oce-0.17 (see https://github.com/tpaviot/smesh)

tpaviot, 2016-07-14:
Latest smesh release should hopefully be reaedy to be wrapped insided pythonocc https://github.com/tpaviot/smesh/releases/tag/6.7

jf---, 2016-07-15:
@tpaviot terrific there's a new SMESH release, awesome!

e-dub, 2016-07-15:
@tpaviot thanks for the quick answer and the effort releasing a new SMESH!  Much appreciated!

SMESH 6.7 builds with perfectly!  Though when atttempting to compile pyOCC with SMESH, I am getting the following error.

Does this have to do with the fix you mentioned in the release notes https://github.com/tpaviot/smesh/releases/tag/6.7

[  0%] Swig source
Scanning dependencies of target _StdMeshers
[  0%] Building CXX object CMakeFiles/_StdMeshers.dir/Unix/x86_64-MinSizeRel-64/StdMeshersPYTHON_wrap.cxx.o
In file included from ~/opt/pyoccNew/build/Unix/x86_64-MinSizeRel-64/StdMeshersPYTHON_wrap.cxx:3810:0:
/usr/local/include/smesh/SMESH_DataMapOfElemPtrSequenceOfElemPtr.hxx:50:23: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_BASECOLLECTION (SMESH_BaseCollectionSequenceOfElemPtr, SMESH_Sequenc

Thanks again!

tpaviot, 2016-07-15:
@e-dub You still have to wait the wrapper to be updated. I'll soon push a branch you can test.

e-dub, 2016-07-16:
@tpaviot 
Thanks!  Let me know, I would be happy to test it out!

sc-ita, 2016-07-25:
@tpaviot When do you think you will be ready with the wrapper? I'm currently working on my master thesis and I would like to mesh a model already built with pythonocc. I was wondering if there was a mesher integrated into that, so if there could be this possibility it would be great to test it! Thanks in advance!

e-dub, 2017-02-14:
Just built and tested pythonOCC with SMESH and, at least with 2D quad meshes, it is working beautifully!

closed 2017-02-14 by e-dub

tpaviot, 2017-02-14:
@e-dub what did u build ? which branch ? os ? compiler ? it's still a work in progress, any feedback is welcome

e-dub, 2017-02-14:
My build is with the following:

OS: Kubuntu 16.04 
SMESH: r6.7.1
OCE: r17.2
pythonOCC: review/smesh-support

A could then build a couple simple surface structures and discretize them with 2D quad mesh.

I am fighting with a 3D mesh that is not giving me the expected results, i.e. not meshing all surfaces in 2D and then being upable to mesh in 3D.  Now I am trying to figure out if it is related to the geometry (a gear created with pyGear), my computer, etc. If interested, I have attached the exported step and resulting mesh as unv. 

[Gear.unv.zip](https://github.com/tpaviot/pythonocc-core/files/773742/Gear.unv.zip)
[PyGear.stp.zip](https://github.com/tpaviot/pythonocc-core/files/773738/PyGear.stp.zip)

tpaviot, 2017-02-14:
@e-dub ok, thank you, I also tested on Ubuntu 16.04. Feel to free to open smesh related tickets, I don't have time to intensively test the behaviour