tpaviot/pythonocc-core #290: pythonocc performance when loading large number of shapes(vertices and faces)

jackshen2018, 2016-07-15:
I am using pythonocc to visualize a mesh data(sets of nodes and elements), I read the node coordinates and element node connections out and convert them to occ points and wire. My code is pasted below:
class hexahedron():
    
    def __init__(self, points):
        self.points = points
        
        self.shape = None
        self.shapes = []
    
    def toShape(self):
        
        vertices = [ gp_Pnt(p.coord[0],p.coord[1],p.coord[2]) for p in self.points ]
        oFaces = []

        builder = BRep_Builder()
        shell = TopoDS_Shell()
        builder.MakeShell(shell)
        
        for face in faces:
            edges = []
            for i in range(len(face)):
                cur = face[i]
                nxt = face[(i+1)%len(face)]
                segment = GC_MakeSegment(vertices[cur],vertices[nxt])
                edges.append(BRepBuilderAPI_MakeEdge(segment.Value()))
                
            wire = BRepBuilderAPI_MakeWire()
            for edge in edges:
                wire.Add(edge.Edge())

            oFace = BRepBuilderAPI_MakeFace(wire.Wire())
            oFaces.append(oFace)
            
            #self.shapes.append(oFace.Shape())
            self.shapes.append(wire.Shape())

I don't know how to insert a code in a post, sorry for the mess. Here are my problems:
#1 pythonocc can only load approximately 1500 hexahedrons and breaks with an ERROR(RuntimeError: StdFail_NotDone)
#2 If I switch to show only wire shapes, everything is ok except that the speed is quite show. My model has about 20,000 nodes and 10,000 elements.

Can anyone tell me how to improve this?
Thanks
Jack SHEN

tpaviot, 2016-07-15:
@jackshen2018 What is exactly the error log ? Actually a StdFail_NotDone is not related to a "performance" issue,  may be the point cloud is not consistent.  Second point, "quite slow" is not accurate enough. Did you measure the time per face or time per wire ? What are your expectation ? Did you run any benchmark with an other tool ? Finally, what is your intent: are you sure you need a brep representation of your mesh? is it for display only or not ? Well, please give a few additionnal information so that the issue can be processed.

jackshen2018, 2016-07-15:
Dear @tpaviot, thanks for your quick reply. I am new to pythonocc, I only got the error of "RuntimeError: StdFail_NotDone;BRep_API: command not done" from the eclipse console. My code and test file is attached below.
About the speed performance, I did not measure a accurate time, but it is too much slow compared with any FEM software I used, and it rendered about 10s when I did some 3D transformation with mouse.
I only need to display my mesh, there are duplicated faces and edges between adjacent elements in my code. Any idea to improve my code is appreciated.  

[test3.txt](https://github.com/tpaviot/pythonocc-core/files/365277/test3.txt)
[mesh.txt](https://github.com/tpaviot/pythonocc-core/files/365276/mesh.txt)

jackshen2018, 2016-07-15:
I do not want the displayed shape to be selectable, how to disable selection when moving cursor to a shape?

jf---, 2016-07-15:
hi @jackshen2018 ,

@tpaviot comment wrt BRep's vs meshes is the key here.
converting a mesh -> BRep is usually a bad idea.
However, if you want to visualize your mesh in OCC / OpenGL, then please take a look at [this example](https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_visualization_graphic3d_custom_opengl.py) 

rather than using expensive BRep data types, you'll use OpenGL more directly, which allows much faster drawing. things like selection are more difficult to handle.

finally, libraries like VTK are much more suited to visualize FEM data

tpaviot, 2016-07-15:
Totally agree with @jf--- pythonocc is not intended to data visualization, dedicated frameworks such as VTK are definitely better. Export a mesh structure to BRep only makes sense if you @jackshen2018 need to perform topological operations on the mesh using BRep algorithms, or advanced geometrical algorithms

jackshen2018, 2016-07-19:
Dear @jf--- and @tpaviot, thanks for your support. I tried VTK according to your advice. It is indeed more efficient in visualizing large number of points and grids.

closed 2016-08-09 by tpaviot