tpaviot/pythonocc-core #291: glueing example for step files?

yoeldk, 2016-08-04:
Hi,

Can one of you please direct me to a code snippet which glues two step files together?
I could find something for IGES, but not for STEP.

Thanks in advance,
Joel

tpaviot, 2016-08-09:
@yoeldk "glueing" STEP files is confusing. You mean one shape per step file, a boolean fuse, then export to a single STEP file ? If you have an example with IGES, it should be quite easy to port to STEP. Please contribute your code here, I'll fix it. I don't have time to create the code from scratch, it will be a good exercise for you.

jf---, 2016-08-09:
@yoeldk , are you refering to [this example?](https://github.com/jf---/pythonocc-core/blob/35ca7359fcec672b3b76ce6fb57710afa7ab66a5/examples/core_topology_glue.py#L151-L151)

yoeldk, 2016-08-23:
@jf---  yes ,can I use BRepFeat_Gluer to glue 2 STEP files (each containing a single object) togeter?

yoeldk, 2016-08-31:
I just saw that I can use BRepAlgoAPI_Fuse so that solves my problem.

closed 2016-08-31 by yoeldk