tpaviot/pythonocc-core #292: Crash due to new memory management

rainman110, 2016-08-08:
The following minimal example demonstrates some problems since we are using the new memory management:

```python
import OCC
import OCC.GC

a = OCC.gp.gp_Pnt(0., 0., 0.)
b = OCC.gp.gp_Pnt(100., 100., 100.)

temp = OCC.GC.GC_MakeSegment(a, b).Value()
line1 = temp.GetObject()
print line1.FirstParameter() #OK

temp = OCC.GC.GC_MakeSegment(a, b)
line2 = temp.Value().GetObject()
print line2.FirstParameter() #OK

line3 = OCC.GC.GC_MakeSegment(a, b).Value().GetObject()
print line3.FirstParameter() #Crash
```

I'll have a look into it and try to understand, what is going wrong.

Obviously, the handle to the trimmed curve (by calling ```.Value()```) gets deleted after line3. Thus, the next line crashes.

Is this valid behaviour? If yes, this is still very unintuitive.

jf---, 2016-08-08:
hi @rainman110 , 

funny, whilst replying to a post on the pyocc forum, I ran into the exact same issue.
IMO, `OCC.GC.GC_MakeSegment(a, b).Value().GetObject()` is sufficiently clear that indeed this problematic for python's garbage collection.

AFAIK, the issue is somewhat contained to the `OCC.GC` module.
The rest of the API isnt (that much) affected by this pattern ( and GC is a pretty thing wrapper over  `OCC.gp` )

My 2 cents; not ideal, but not critical.

---

\<pedantic>
btw, just in terms of style

```python
from OCC.gp import gp_Pnt
a = gp_Pnt(0., 0., 0.)
```

is easier on the eyes 
\</pedantic>

rainman110, 2016-08-08:
hi @jf--- 

in fact, I copied this example directly from an email I got from Lense Swaenen. Maybe the forum entry was also from him. Hence, the style :smirk: 

The issue does not only exist in ```OCC.GC```. It's a generaly issue, whenever a C++ function/method returns a Handle.

I already implemented a fix for it: https://github.com/rainman110/pythonocc-generator/commit/30e42df10eb7517fe17ff8c08b9caea8388a69c0

I'll now create a PR.

jf---, 2016-08-08:
My goodness, @rainman110 , you manage memory as a true zen master you are...

[image]

Lense's question is interesting, in fact it turned out more tricky than it ought to to come up with an elegant solution. More later.

tpaviot, 2016-08-27:
Fixed by PR #295

closed 2016-08-27 by tpaviot