tpaviot/pythonocc-core #296: STL performance and resolution

Puffton, 2016-08-17:
**STL Reader:**
I'm working with larger STL files (commonly binary files sized 100 MB - 2 GB). ANSA can deal with these files quite swiftly, but pythonOCC doesn't seem able to read them in reasonable time. After waiting 3 hours (python running at 100% CPU on a single process) to read a 200 MB STL I gave up. Are there known issues with STL reader performance? This is using the example code.

**STL Writer:**
Is it possible to control the sampling resolution for CAD files when they are saved as STL? Right now it seems to be done entirely automatically?

jf---, 2016-08-18:
hi @Puffton ,

So, the OCC.STL module is a little awkward. What is does is that every triangle becomes a nurbs patch. Hence the very high cost of importing STL meshes. Now, [SMESH](https://github.com/tpaviot/pythonocc-core/pull/53/files) offers an import such that the mesh is merely rendered in the viewport, and not reconstructed as a BRep type.

Personally, I've been using an approach where I use pyassimp while pushing the imported mesh to opengl [like so](https://github.com/tpaviot/pythonocc-core/blob/master/examples/core_visualization_graphic3d_custom_opengl.py)

So in order to help you a little more precise its relevant to know what you're aiming for.
Do note that pythonocc's focus is on BRep geometry, not meshes...

Puffton, 2016-08-18:
Thank you for your quick reply @jf---! Currently I'm using pythonOCC to load files (IGES, STEP and STL) and outputting a scaled binary STL mesh. I'm not rendering anything onto the screen. Later me and my colleagues might want to use more meshing capabilities such as volume mesh creation from a clean surface mesh. Perhaps SMESH is the way to go for us then?

jf---, 2016-08-18:
ah, OK I understand the problem a lot more precise now.
- not suprised about the poor STL performance... 
- possible [numpy-stl](https://pypi.python.org/pypi/numpy-stl) is more convenient if the operation is pretty trivial
- yes, so volumic meshing is what SMESH is all about
- here's an interesting post on [SMESH ](http://www.pythonocc.org/resources/meshing/pythonocc-and-smesh/) and [old one](http://www.pythonocc.org/news/pythonocc-as-pre-processor/) here
- the SMESH lib was recently updated, and the python wrappers need an update

Puffton, 2016-08-29:
Okay, thanks a lot for your help! Oh, and it would be cool if the precompiled binaries fetched with Conda included SMESH too (at least as an option, if that is possible)!

tpaviot, 2017-06-06:
@Puffton The current master branch wraps smesh, an example of fast STL loading is here f8f3735

tpaviot, 2017-06-07:
@Puffton can u please point me to a big STL file I could test the fast STL example ?

tpaviot, 2017-06-10:
@Puffton With the current pythonocc-core master branch, based upon oce-0.18/occt6.9.1, you can set the STL file resolution by setting the mesh deflection value. I updated the STL example to show how to modify this parameter in order to change the STL file resolution see commit bb131893b20ce7e8a9af2a6554e877b92f421e72

tpaviot, 2017-06-10:
Both STL performance and resolution issues are solved, closing this issue.

closed 2017-06-10 by tpaviot