tpaviot/pythonocc-core #298: Zoom in on a part of a model

lalvarezt, 2016-08-29:
Hi,
I'm trying to zoom in on a part of a model, for that I calculate the projection of a given point in the 3D space on the screen (using Project), but the result of Project gives me the same point in 3D space not the required pixel on the screen required to do the zooming.

Here's the code

xmin, ymin, zmin, xmax, ymax, zmax = self.motor.rotor_boundingbox()
xpmin, ypmin = self.OCE_widget.display.View.Project(xmin, ymin, zmin)
xpmax, ypmax = self.OCE_widget.display.View.Project(xmax, ymax, zmax)
self.OCE_widget.display.ZoomArea(xpmin, ypmin, xpmax, ypmax)

I tried another approach, using the camera in the View

self.OCE_widget.display.View.Camera().ConvertWorld2View(gp_Pnt(xmin, ymin, zmin))

But it says that the camera handle object doesn't contain the function ConvertWorld2View.

Am I doing something wrong, or there is another way to accomplish the same task.

Thanks

jf---, 2016-08-29:
well, Shift + right mouse button does provide the functionality of zooming into an area, see the [ZoomArea](https://github.com/tpaviot/pythonocc-core/blob/master/src/addons/Display/OCCViewer.py#L527) method

or are you trying to zoom in on a selection in your viewer?

lalvarezt, 2016-09-09:
sorry for the late reply and thanks for your answer, yes indeed the ZoomArea does provide the posibility that I want, also using WindowsFit. I translated the point coordinates to pixel coordinates and did the Zoom in. But i would like to access the camera to do transformations on it.

jf---, 2016-09-09:
@lalvarezt, that's a coincidence, I share the need for being able to access the camera, I just reported #301 to do exactly that...

jf---, 2016-12-13:
OCCT 7 supports this functionality

tpaviot, 2017-12-01:
@lalvarezt The Camera object is available in pythonocc since the 0.18 release

@jf--- OCCT 6.9 already supports the ConvertWorld2View method, and is available in pythonocc

Committed an example related to 3d to 2d screen coordinates see 44e63d607a0f410f40f96ec03e46429ecb750823

closed 2017-12-01 by tpaviot