tpaviot/pythonocc-core #299: More complex GUI

yoeldk, 2016-08-31:
Hi,

Any code snippet with a more complex GUI (not SimpleGUI)?

Thanks,
Yoel

jf---, 2016-08-31:
I've been working a lot on a fairly involved QML gui.
This allows for HUD widgets for example.
However, requires OCC 0.18 

So for now, no, but something is cooking ;)

yoeldk, 2016-09-01:
@jf Thanks, sounds good..keep me updated

jf, 2016-09-01:
@jf---, I am opening a support ticket with GitHub to see what can be done here, but pls, do me a favour and correct these wrong username references whenever you see that happen? I haven't seen you do any.

@yoeldk, pls reference the correct username. Thank you.

yoeldk, 2016-09-05:
@jf--- Sorry and thanks for notifying me, I haven't noticed I actually referenced the wrong user until you asked me to fix it.

So just to understand, I'm currently unable to use pythonOCC with a GUI that I write on my own (say using pyqt or some other library)? Do I have to use SimpleGUI in order to render 3d objects?

Thanks,
Yoel

jf---, 2016-09-05:
- nope, its actually easy to use the pyqt widget in your own app, its a good start...
- SimpleGUI is a good start for creating your own app / render 3d object, just take a look at the examples ( the question is fairly non-specific, sorry ) 
- inherit from [qtViewer3d](https://github.com/tpaviot/pythonocc-core/blob/master/src/addons/Display/qtDisplay.py#L94) and off you go ;)

yoeldk, 2016-09-08:
@jf---  regarding your second point, I managed to render 3d objects using SimpleGui, I was just wondering if there are examples showing how can I modify it except from adding menus(i.e adding pictures/widgets to it) or alternatively examples that demonstrate a usage in a "generic" GUI that allows more flexibility

jf---, 2016-09-08:
This is [WIP based](https://github.com/jf---/aoc-xchange/commit/4904330f441a139b4911336b52a0fe2601d35e3b) on Doug Blanding post recently...
Not a great example, since still WIP, but perhaps that does that point you in the right direction?

yoeldk, 2016-09-12:
@jf---  I'm  newbie to CAD/pythonOCC. I guess I'm looking for a very simple example where a 3d object is rendered on a GUI that was created using a "generic" GUI library that will allow me to modify it as I will. Maybe qtViewer3d is what I need, just looking for a simple example that shows how it renders a OCC.TopoDS.TopoDS_Shape

jf---, 2016-09-12:
@yoeldk , yes, it'll be a good start for sure. Qt has a pretty easy learning curve. Personally, I'm really fond of QML.

jf---, 2016-09-27:
@tpaviot , too broad, can you close this?

closed 2016-09-27 by tpaviot