tpaviot/pythonocc-core #301: missing Graphic3d.Camera / Graphic3d.ClipPlanes methods

jf---, 2016-09-09:
these methods seem missing:

```python
In [25]: dir(display.View.GetClipPlanes())
Out[25]: 
['__class__',
 '__cmp__',
 '__delattr__',
 '__doc__',
 '__eq__',
 '__format__',
 '__ge__',
 '__getattribute__',
 '__gt__',
 '__hash__',
 '__hex__',
 '__init__',
 '__int__',
 '__le__',
 '__long__',
 '__lt__',
 '__ne__',
 '__new__',
 '__oct__',
 '__reduce__',
 '__reduce_ex__',
 '__repr__',
 '__setattr__',
 '__sizeof__',
 '__str__',
 '__subclasshook__',
 'acquire',
 'append',
 'disown',
 'next',
 'own']

In [27]: dir(display.View.Camera())
Out[27]: 
['__class__',
 '__cmp__',
 '__delattr__',
 '__doc__',
 '__eq__',
 '__format__',
 '__ge__',
 '__getattribute__',
 '__gt__',
 '__hash__',
 '__hex__',
 '__init__',
 '__int__',
 '__le__',
 '__long__',
 '__lt__',
 '__ne__',
 '__new__',
 '__oct__',
 '__reduce__',
 '__reduce_ex__',
 '__repr__',
 '__setattr__',
 '__sizeof__',
 '__str__',
 '__subclasshook__',
 'acquire',
 'append',
 'disown',
 'next',
 'own']
```

tpaviot, 2016-09-09:
It's available in the current master. See https://github.com/tpaviot/pythonocc-core/blob/master/src/SWIG_files/wrapper/Graphic3d.i#L2242

jf---, 2016-09-09:
Aiii! Sorry for the noise...

jf---, 2016-09-09:
For what's it worth, I've integrated the [cura slicing engine](https://github.com/Ultimaker/CuraEngine) with pythonocc

[image]
[image]

tpaviot, 2016-09-10:
@jf--- interesting, maybe you could post this to another location

jf---, 2016-09-27:
hi @tpaviot , so the `display.View.Camera()` object is still a SwigPyObject / not wrapped in 0.17. 
the same is true for `display.View.GetClipPlanes()`

```python

from OCC.BRepPrimAPI import BRepPrimAPI_MakeBox
from OCC.Display.SimpleGui import init_display

display, start_display, add_menu, add_function_to_menu = init_display()

# Create a box
s = BRepPrimAPI_MakeBox(200, 100, 50).Shape()

display.FitAll()
cam = display.View.Camera() # SwigPyObject
plns = display.View.GetClipPlanes() # SwigPyObject

start_display()
```

tpaviot, 2016-09-27:
@jf--- A typo in the SWIG generator prevents some Graphic3d_ classes to be wrapped. Fix is on the way

jf---, 2016-09-27:
<<< Cool >>> ! 
especially looking fwd to the Shader class 🎩

tpaviot, 2016-09-28:
PR#327 adds Graphic3d.Camera, Graphic3d.ClipPlanes is still missing

lalvarezt, 2016-10-31:
@jf-- @tpaviot Hi, I think I've found the reason for the missing classes, first, the clipplane is not being generated because is declared as excluded on modules.py in the pythonocc-generator, after fixing this there is another issue, the generator is not handling the typedefs for the particular classes (in this case is the Equation typedef), only the global ones, I've fixed the generator and now the wrapper for the classes are correctly generated and the code compiles without issues, I'll upload the code, but first I've got to do some testings.

jf---, 2016-10-31:
@lalvarezt sounds very promising!
a PR in this would be much appreciated, we look fwd to it 👍

tpaviot, 2016-11-01:
@lalvarezt Some classes/methods are excluded from the wrapper because they cause compilation failures. You're right currently only global typedefs are generated, it's a good thing if you can handle specific class typedefs. Feel free to create a PR on pythonocc-generator

lalvarezt, 2016-11-01:
@tpaviot @jf--- here is the code that adds the missing typedefs, as far as a checked there are only 2, one is Equation from the clipplanes class and the other one is Target from the classes RepBuilderAPI_VertexInspector, BRepMesh_CircleInspector, BRepMesh_VertexInspector, I think this is related to the class NCollection_CellFilter class that is also excluded.
here's the commit 1ff1588473bc2ec57d0c5582c502d16dcf4a9867

closed 2016-11-02 by tpaviot

jf---, 2016-12-22:
hi @tpaviot , so I've made a `Graphic3d_Camera` example, that said, `plns = display.View.GetClipPlanes() # SwigPyObject` returns an unwrapped pointer

PR #349 (2016-11-02): class typedef fix
Allows Graphic3d_ClipPlane to be available. Thanks @lalvarezt Fixes #301